nu-radio/NuRadioMC #1: Number of ray tracing solutions is not stable as a function of root finding tolerance parameter

clark2668, 2018-09-17:
The number of solutions for the ray tracing problem is not stable as a function of the root finding tolerance parameter `tol` in `analyticraytracing.py`.

For example, the following code:

> x1 = np.array([478., 0., -149.])
x2 = np.array([635., 0., -5.])  # direct ray solution
ice = medium.ARAsim_southpole()
r = ray.ray_tracing(x1,x2,ice)
r.find_solutions()

Returns only one solution when `tol=1e-4` but has two solutions when `tol=1e-5`.

anelles, 2019-05-23:
@cg-laser does your improvement in pull request #80 also address this issue?

clark2668, 2019-05-23:
@cg-laser , no, it doesn't. I identified this problem back when I was doing a comparison of the python vs c++ ray tracer. the C++ ray tracer was finding a second solution when the python one wasn't, and I found that if I pushed the tolerance parameter tighter in the python implementation, it could find the second solution. I suspect something in the scipy root finding algorithm maybe?

cg-laser, 2019-05-23:
Maybe, I can check. The objective function should look better now.

cg-laser, 2019-05-23:
why is @clark2668  comment not visible on github?

clark2668, 2019-05-23:
Sorry @cg-laser , I deleted the comment. I realized I misunderstood the question before I answered.

fschlueter, 2025-05-02:
Can that be closed?

clark2668, 2025-05-05:
Oh, given that it's ancient, seems like yes haha?

closed 2025-05-05 by fschlueter