nu-radio/NuRadioMC #61: Most examples in SignalGen class are broken

clark2668, 2019-04-19:
Many of the tests in the SignalGen class rely on calls like: 

`import NuRadioMC.SignalGen.parametrizations as param`
`spec4 = param.get_frequency_spectrum(E, theta, ff, em, n_index, R, model='ZHS1992')`

But the `get_frequency_spectrum` method was removed, so these no longer work.

anelles, 2019-05-20:
I suggest that we write new tests. @cg-laser probably the figures you made for the paper could replace these tests, no?

anelles, 2019-07-12:
I would suggest to move these to the general tests and remove all broken ones. As in #106 we are overhauling the whole testing anyway.

anelles, 2019-08-27:
We are testing the SignalGen examples now on the built-bot. Older examples may no longer work.

closed 2019-08-29 by anelles