nu-radio/NuRadioMC #65: synchronization between NuRadioMC and NuRadioReco

cg-laser, 2019-04-25:
Some changes in NuRadioMC require changes in NuRadioReco, e.g. saving of additional parameters in the parameter storage. How do we synchronize these changes? 
Can we automatically check for a NuRadioReco version? But right now we don't really habe a versioning. 
Can we read out the git commit status? And give a warning or stop if NuRadioReco is not at least at a certain commit level?

anelles, 2019-04-29:
I think versioning is in principle good pratice anyway, the question is when to start this. Are we at a stable enough state to start this now?

cg-laser, 2019-05-17:
I investigated versioning a bit more. One easy way is to store the unique status of the git repo (hash of first line of `git show`). 
In addition we should start thinking about proper versioning. See (https://git-scm.com/book/en/v2/Git-Basics-Tagging) for how to tag versions. One policy could be that we always increment minor versions with each pull commit to master (including a descriptive message). In this way one could get a quick overview of what changed between two versions.

christophwelling, 2019-05-17:
I think the main problem will be how to tell if a change in NuRadioReco affects NuRadioMC and vice versa. Once we have the framework properly implemented, most commits in NuRadioReco will probably not affect NuRadioMC, so specifying the commit is probably overkill. But I can't think of anything better than manually specifying a minimum required version.
Will doing this with git commits work if someone is working on a branch? Printing a warning if versions don't match should be fine, but throwing an error would be super annoying.

anelles, 2019-05-17:
This really triggers the thought about continuous integration again. We need tests running, otherwise, this will be an endless problem.

anelles, 2019-05-17:
I will have a look at what you can go with Travis CI which github seems to promote.

cg-laser, 2019-05-17:
the git commit hash was also more to be able to reproduce a result. This hash uniquely defines the state of the repository (if not uncommited local changes were present.)

cg-laser, 2019-05-17:
It seems to me that after the time changes from Christoph, that NuRadioReco is getting pretty stable. The same for NuRadioMC. At least I can't think of any major interface/framework change. Or do you @anelles @christophwelling @daniel-zid  have anything else in mind? 

It would also be good to release a version 1.0.0 for the papers and the DESY workshop. 
I suggest the following scheme: Bump up the sub-sub version after each pull request merge. 1.0.1, 1.0.2 etc. 
Bump up sub version 1.0 - 1.1. if substantial new features are implemented (that might even break backward compatibility). This complies with pep8 standards: https://www.python.org/dev/peps/pep-0440/#final-releases

The next question is where to store versions: I think the standard is 
` __version__ = "1.0.1"`
below the doc string of the main __init__.py. Then 
```
import NuRadioMC 
NuRadioMC.__version__
```
should print the correct version.

daniel-zid, 2019-05-17:
It sounds reasonable to me. And I think that what we have now, we should
release a 1.0.0 version.

Any new thoughts on migration to Python3? The warp machines spit some
future warnings from
time to time. Am I the only one to have this issue?

christophwelling, 2019-05-17:
I thought issue with Python3 was that SnowShovel still needs an older ROOT version. Are there any updates on that? If that is fixed, I think we should switch as soon as possible (and before version 1.0).
We still need to implement our coordinate system properly. Right now most of it is not proprely defined and just implied.

daniel-zid, 2019-05-17:
Yes, yes, I know. What I meant is that I get some future warnings while
using Python2,
urging me to use Python3 syntax. Am I the only one?

/afs/
ifh.de/group/radio/software/anaconda5.3/lib/python2.7/site-packages/scipy/signal/signaltools.py:2233:
FutureWarning: Using a non-tuple sequence for multidimensional indexing is
deprecated; use `arr[tuple(seq)]` instead of `arr[seq]`. In the future this
will be interpreted as an array index, `arr[np.array(seq)]`, which will
result either in an error or a different result.

christophwelling, 2019-05-17:
Yes, I get that too. Also on my own computer.

cg-laser, 2019-05-17:
> This really triggers the thought about continuous integration again. We need tests running, otherwise, this will be an endless problem.

We actually have a simple validation test. Just execute ./validate in the test folder. Currently it fails, likely because of your changes to the cross section. But it's difficult to check this afterwards. So yes, setting up a continuous testing would be great.

cg-laser, 2019-05-18:
> I thought issue with Python3 was that SnowShovel still needs an older ROOT version. Are there any updates on that? If that is fixed, I think we should switch as soon as possible (and before version 1.0).

yes this is still the case. I just tested it, see my email to corey. Feel free to test/fix it. Anna can tell you how to access snowflake. I already created a test at `/home/arianna/software/snowshovel_root6/test`
But I don't think we need python3 for 1.0. Python 3 can also be our 1.1 release. And even if we say we want 1.0 for python 3, we can still start versioning with 0.9 or so

anelles, 2019-05-20:
Just to catch up here: I am all for releasing a 1.0.0 now. After having cleaned up all the small things. Should we aim for a specific date?
I don't think Python 3 is needed for now. As usual in science, Python 2.7. will stick around a bit longer, so having a release for Python 2.7 is probably even good. (Despite not wanting people to encourage this)
About continuous integration, I was also thinking about running some examples. If I understood correctly, it is also about things not breaking, rather than checking whether the result is the same.

cg-laser, 2019-05-20:
yes we need both. Not breaking anything is the first step. But we need also make sure that our physics output remain the same. And this fast test run does both. It runs a standard simulation, so it also tests . most things. And only takes a few seconds. This is actually the quickest thing I can think of to test things.

anelles, 2019-08-29:
I think we are doing this already with all our tests. We even discussed whether this creates too much intedependecy. So well done everyone ;)

closed 2019-08-29 by anelles