nu-radio/NuRadioMC #74: Substituting YAML for JSON

anelles, 2019-05-15:
JSON is not ideal for a detector description since it doesn't allow comments. We are using YAML for the config files, which allows comments. 
It may be worthwhile moving to one type in the future. However, TinyDB is only compatible with JSON, so in order to keep both the database support and a human-readable format, a different solution is needed to change this. 
Not a high priority problem, but should be kept in mind, should changing from TinyDB at some point or if someone has a good idea.

anelles, 2019-07-12:
#108 is probably related to this.

fschlueter, 2025-05-02:
Do we want to close this?

fschlueter, 2025-08-07:
@anelles this is still very much open I guess. I wonder though if there is a majority to get rid of tinydb all together.

closed 2025-10-15 by fschlueter