nu-radio/NuRadioMC #97: gLog value in analytic ray tracing

daniel-zid, 2019-06-24:
The logarithms for l1 are -inf sometimes. This tends to happen with the Greenland refractive index model and for refracted solutions. Since for Greenland we expect less refracted solutions due to the ice properties, I ignore these (relatively few) cases. I am going to push this workaround to a new branch.

cg-laser, 2019-06-24:
@bhokansonfasig maybe you can provide more insights. @daniel-zid is finding problems using the analytic solution for the travel time/distance with the Greenland n(z) profile. I think the relevant part of the code is: https://github.com/nu-radio/NuRadioMC/blob/master/NuRadioMC/SignalProp/analyticraytracing.py#L282-L285
@daniel-zid maybe you can provide more insights why l1 is zero. E.g. for which depth is this happening.

cg-laser, 2019-06-24:
See also https://github.com/nu-radio/NuRadioMC/pull/100 for the pull request to fix it.

daniel-zid, 2019-06-24:
It happens when I use the Greenland refraction profile and I set an array with antennas at -48.5, -49.5, -50.5 and -51.5 meters. I haven't checked if that happens for more depths but in principle it can. I don't have right here more data for a particular event that crashes but I can have some data tomorrow.

bhokansonfasig, 2019-06-24:
I remember running into this in my implementation as well. The problem is when the index of refraction at the endpoint depth ends up being equal to the asymptote due to numerical precision (i.e. when you're very deep so n(z)=1.78 for South Pole ice). I had to rework my ray tracer to handle cases of deep ice with an approximation of uniform index in these regions, which came with some additional overhead.

If you check [here](https://github.com/bhokansonfasig/pyrex/blob/f0437339ed82b1d3fe5ffbb0664ddf60d11a5f61/pyrex/ray_tracing.py#L857-L864) you'll notice that when the "deep" condition is met, different equations are used to calculate the integrand. Similar conditions were also necessary in other parts like the time of flight. Then integration then needs to be split up into integration in the exponential region and integration in the uniform region (as done [here](https://github.com/bhokansonfasig/pyrex/blob/f0437339ed82b1d3fe5ffbb0664ddf60d11a5f61/pyrex/ray_tracing.py#L585-L596)).

Attached is a document outlining my ray tracing technique, and section 4.5 discusses this issue (I've been meaning to add this document to the PyREx documentation, but haven't gotten around to it yet):
[Analytical Ray Tracing.pdf](https://github.com/nu-radio/NuRadioMC/files/3322435/Analytical.Ray.Tracing.pdf)

bhokansonfasig, 2019-06-24:
Oh, actually maybe this is already done in NuRadioMC? At least it appears in the python implementation

daniel-zid, 2019-06-24:
It is implemented but it still crashes. I don't know why, I haven't looked at it in detail.

cg-laser, 2019-06-24:
Maybe the deep condition needs to be modified as the firn is only 100m deep for Greenland. Currently, `z_deep = -500 * units.m`. @daniel-zid could you check if it works when you set `z_deep = -250 * units.m` or so. If this solves the problem, we need to come up with a way to determine the 'deep' threshold on the fly from the n(z) profile.

bhokansonfasig, 2019-06-24:
Oh, yes this may be the problem. I set my depth as the depth where the index of refraction reaches some fraction of the asymptote, since this is a numerical rounding issue in the index of refraction at its heart

anelles, 2019-06-25:
I guess it is really the firn depth that is varying and causing the problem. The firn is much thinner in Greenland. Either an on-the-fly calculation or another parameter that is part of the ice model. I think the latter might be easier to implement, but not as elegant of course.

daniel-zid, 2019-06-25:
z_deep = -250 * units.m works well for Greenland. It should be easy to implement a z_deep as a function of the refractive index using the medium as an argument.

daniel-zid, 2019-06-25:
I've added a function to get the z_deep parameter for different profiles. It calculates the depth for which the relative refractive index difference is 0.035% and assigns it to z_deep. For Greenland, this depth is -249.858 m, and for the SouthPole 2015 model is -502.084 m. See pull request #100.

closed 2019-06-28 by daniel-zid