nu-radio/NuRadioMC #132: Veff calculation in utitlies

cg-laser, 2019-09-11:
I would like to change how the effective volume is calculated. Currently the "water equivalent effective volume steradian" is calculated. Which is the original effective volume x Omega (typically 4pi) x rho_water/rho_ice. This definition has rather historic reasons than being very useful. It is also error prone because one needs to remember to get the interaction length of neutrinos in water to convert to a limit, and for the effective area one needs to remember the Omega of the simulation. 
I would like to change it to just give the effective volume (for ice). 

And we actually got it wrong in the limit calculation. We calculate the interaction length for ice, but the effective volume was converted to 'water'. So we're actually10% better...

Because changing the utility script will change the results of everyone using it, I wanted to discuss it first before changing anything. What do you think?

daniel-zid, 2019-09-11:
I don't think I follow. What is currently calculated is not

effective volume x Omega (typically 4pi) x rho_water/rho_ice

but rather

effective volume x Omega (typically 4pi) x rho_ice/rho_water

According to what we had discussed, the cross sections are given for a medium with the density of water, so we need to correct for the density. I never understood why. I agree that the cross section is medium-dependent, but it is not density-dependent, and when we calculate the interaction length we put the right medium density. That's why I proposed to remove this ratio long ago.

By the way, a heads up. There's an error on the Veff module, it does not return the correct Veffs (I forgot to include an append in one of the last commits). Could you change it?

daniel-zid, 2019-09-11:
I don't think we have converted anything to water, or if we have done it, we have done it wrong. We have 'corrected' the water volume to ice volume, but that does not make sense, at least the way I understand it. Therefore, we should remove the ratio and use the neutrino cross section for water. It doesn't matter the density or the state as long as it is water - cross sections for ice and water should be the same.

Right after the Veff is calculated, the following line should be:
'Veffs[trigger_name].append(Veff)'
Could you re-implement it on the master branch?

cg-laser, 2019-09-11:
I already fixed it my branch. 

Isn't the correct way to calculate the effective volume:
Veff = V / N * sum over triggered events (weights_i) and divide it by the interaction length for 'ice' to get the effective area?

daniel-zid, 2019-09-11:
Yes, but that's not what we have now. We have a density ratio that makes no sense.

cg-laser, 2019-09-11:
yes exactly! So you opt for changing the Veff utility script?

daniel-zid, 2019-09-11:
Sure. I don't know why it's taken so long.

I don't necessarily agree about removing the solid angle, though. If we use a restricted zenith window, as I do all the time for Gen2, it is a pain not to have it there. You should always use the solid angle in the end. Even if you use the whole sphere, you have to remember that you used a solid sphere. In any case I would include the solid angle for each zenith band in the final json file.

daniel-zid, 2019-09-11:
Then, do we put a switch? The solid angle must be there for zenith-dependent simulations one way or another. I have no strong preference but it must be somewhere.

anelles, 2019-09-25:
Do we need to revisit this after the conversation in the inicemc call, @cg-laser ?

cg-laser, 2019-09-25:
No I don't think so. I added a 'conversion to water equivalent' function, and extensive comments. I think we can close this issue. (see also #133 ).

closed 2020-06-29 by cg-laser