nu-radio/NuRadioMC #139: automatically simulate efield and noise adder?

cg-laser, 2019-09-18:
Currently the user needs perform the efield to voltage conversion, downsampling and noise adding. These steps are so general that we could also perform them always in the main simulation.py script. 
This would have the advantage that we can code noise adding correctly, including adjusting the normalization correctly. 

The downside would be more advanced detector simulations. E.g. for the deep learning data set we first simulate a trigger on the signal only to see where the signal would trigger. Then we add noise and resimulate the trigger. Something like this would still be possible by setting the 'add noise' config to false, but then add noise in the detector description anyway. 

In the future we might want to add noise even before applying the antenna response. This would be precluded from such a change.

My current feeling is to rather leave it as it is.

christophwelling, 2019-09-24:
We could also just wrap these modules up inside another module. We could have something like a standardDetectorResponseSimulator that just runs all the necessary modules with the correct settings.
This would also have the added benefit of making NuRadioReco and NuRadioMC easier to use for beginners. Anyone new can just use the wrapper module and will get a good simulation without having to worry about what modules are needed and with what settings.
Anyone who wants to do a more fancy detector simulation can still just put in all the modules individually like we do now.

cg-laser, 2019-09-24:
yes, this sounds like a good plan

anelles, 2019-09-25:
I like the idea of wrapping the default modules. It does require some work, so who is volunteering?

cg-laser, 2020-06-29:
Implemented in #208

closed 2020-06-29 by cg-laser