nu-radio/NuRadioMC #151: Timing calculation for LPM showers

anelles, 2019-10-15:
As @christophwelling found, there are some strange timing offsets in LPM showers. 
[image]

We discussed that this may be related to the fact that we calculate the ray tracing from the vertex instead of the shower maximum. It may also be possible to have scenarios in which subshowers interfer. What is everyone's thought about this? Could we simulate this better?

daniel-zid, 2019-10-15:
The problem is that first the ray tracing is done to check if the event is seen or not and then the ARZ module is called. Perhaps we could draw a shower profile if the ARZ model is chosen, pass the maximum to the ray tracing method as starting point, perform the ray tracing, and then use the same shower profile to calculate the electric field if the ray hits the detector?

cg-laser, 2019-10-25:
I'm pretty sure that the ray tracing from the start of the shower is not the problem here. In case of an homogeneous medium, the times are correctly calculated within the ARZ module, and the origin of the CS (the point from which we do the ray tracing) doesn't matter. 
So a few ideas how to make the simulation better

1. Do the simulation in a homogeneous medium (either define a new ice model or have the antenna be pretty deep and only use direct rays)
2. simulate HAD and EM showers separately. Use the config parameter `signal.shower_type` so control it. This way you can make the plot separately for EM and HAD showers.

daniel-zid, 2019-11-04:
In the case of an homogeneous medium, times are correctly calculated with the ARZ module, yes. But in the case of a non-homogeneous medium, we can only guarantee that the times for the field coming from the vertex are going to be correct after ray tracing from the vertex. To get the right times for the shower maximum emission, we should trace the rays from the shower maximum. The study you made for the NuRadioMC paper review shows that.

There's the underlying issue that the ARZ formula only works for an homogeneous medium, so if the shower is long enough so that the index of refraction is varies along the shower, the times are not going to be correctly calculated. We should use the n_index for the shower maximum and the ray tracing from the shower maximum. Slicing the shower and calculating the ARZ field together with a ray-tracing for each slice should prove that this approach is more accurate than the one we are using now.

cg-laser, 2019-11-04:
I agree, but these are all corner cases that are not so relevant for a full MC simulation. The likelihood of a vertex within the firn is just very small. 
My guess is that Christoph will get the same plot for a homogeneous medium. This is something I would try out first.

fschlueter, 2025-08-07:
Is this still open? Can it be fixed with starting the ray-tracing from the shower maximum? Christian suggested to first calculate the shower profile and than run the ray-tracer. Is that practical since ARZ is so time consuming? Would it be easier to re-run the raytracer?