nu-radio/NuRadioMC #159: Average index of refraction for SignalGen module

daniel-zid, 2019-11-22:
After talking to Krijn, I'm convinced that the refractive index at the origin cannot be used for calculations. The coherence at the observer position is determined by the observation times, and those are defined by the average index of refraction along the path. If I apply my "spherical ball in the far field" argument to the atmosphere and use the refractive index at the origin I don't get at the correct ZHAireS ZHS formula, which uses the average index. The change to average index will affect parameterisations and the ARZ formula, modifying the Cherenkov cone position, hopefully by not-that-much.

If we want to use the average index, however, a little restructuring of the code must take place. We need to calculate it upon ray tracing and then feed it to the SignalGen module. Two matters must be addressed.

1) Do we want average index (as you assume, I would say yes)?
2) What is the best way to implement it?

cg-laser, 2019-11-22:
I'm not yet convinced that this is definitively the right thing to do. We should also consult with Jaime and Enrique. They must have thought about it before. 
In terms of implementation it should be easy. The medium class already contains a function that returns the average index of refraction between to depths. Hmm but we probably need the average index of refraction along the path? So we would need to add another line integral to the ray tracing code. Also no big deal. As the ray tracing is done before the signal generation, it's also easy to pass it to the signal gen module. 
But I first would like to see the effect. @daniel-zid what one could do quickly is to just call the Askaryan module with n=1.78 and the average index of refraction for a surface station using `get_average_index_of_refraction` from medium.py

daniel-zid, 2019-11-22:
> I'm not yet convinced that this is definitively the right thing to do. We should also consult with Jaime and Enrique. They must have thought about it before.

I doubt they have given it a lot of though for ice, but we can ask.

> In terms of implementation it should be easy. The medium class already contains a function that returns the average index of refraction between to depths. Hmm but we probably need the average index of refraction along the path?

That's what I meant, along the path. That's what gives the correct arrival times.

> So we would need to add another line integral to the ray tracing code. Also no big deal. As the ray tracing is done before the signal generation, it's also easy to pass it to the signal gen module. [...]

True. That's something that should be checked.

fschlueter, 2025-08-08:
I guess this is still very much open?

cg-laser, 2025-08-28:
This was studied recently extensively in the PhD thesis of Dieder Van den Broeck. It is indeed the right thing to do to use the index-of-refraction at the origin (the position of the shower). I suggest to close it. We can anyway now verify this with Corsika8.