nu-radio/NuRadioMC #168: Event IDs for showers from same particle

christophwelling, 2020-01-16:
We want to implement multiple showers or triggers coming from the same particle, e.g. a double bang event or DnR causing 2 separate triggers.

From the reconstruction/NuRadioReco side, the plan is to let these events have different ids and draw the connection by storing the IDs of the associated events in the event class. We then give the event reader a function that returns the associated events belonging to some given event.

From the simulation/NuRadioMC side, this is a bit more tricky. The solution in the HDF5 files is to give events like this the same ID. This is not a good solution, since we want to give them different IDs in the .nur files (and the eventwriter will refuse to write events with the same ID into the same file) but still be able to identify the same events in the .nur and HDF5 files.

Our idea was to add a field "particle ID" (or some better name we come up with) to the HDF5 files. This ID can be the same for events caused by the same particle so we know which events belong together while the eventID is different for each event and is the same as the one written into the .nur files.

Does anyone have any reservations about this plan?

cg-laser, 2020-01-16:
I discussed a similar extension a while ago with Dave Seckel. Each event consists of an arbitrary number of 'sub events'. E.g. one event could be all the particles coming from the first 10 interactions. This would allow do also model the LPM effect and easily implement new physics by changing the event generator. Our idea was to just do this internally in NuRadioMC and merge all sub events back to a master event at the end of the simulation and only save this event. But we can also just save all the individual events. 

Why don't just use the "run_number" as the NuRadioMC "event_id" and "event_id" as the NuRadioMC "sub event id"? So far we ignore the NuRadioReco run_number field completely (and always set it to 0). This way we don't need to touch NuRadioReco at all. 

The technically more difficult part that I see is when to split up events. The current logic is to create one big efield trace with all the signals and then run a trigger module (that then only selects the first trigger and cuts the trace). The simple solution is to just write a new trigger modules that can "double trigger" events. For the second trigger it would create a copy of the event, increase the event number by 1 and cut the traces to the position of the second trigger. 

However this does not necessarily select the second DnR trace. If you want to do this explicitly we would need to change the internal logic to simulate different ray tracing solutions as different events. But I don't think that this is what we want. In reality we don't have this situation. And for several geometries the timing of the two solutions are close together so that we want them in one trace and interfere with each other. 

Ideally we would implement a logic that separates different ray tracing solutions (or contributions from different interactions) into separate events when the time difference is larger than some value. Especially for Moores Bay (and probably also for a deep detector) this would result in speed improvements, as having a looooooong trace slows down the simulation. I tried to do that a while ago but it was easily doable using the current looping structure... so I gave up (but it was also close to the proposal deadline, so I couldn't invest too much time...)

christophwelling, 2020-01-16:
I don't think using the run_number and event_id that way is a good idea. For 2 reasons:
1. In the real data, run_number and event_id will be set by the station, which of course has no idea if two events are related. We want to be able to stitch two events together later without changing run_number and event_id, so we need some other way to mark relations between events.
2. That would mean event_id and run_number can mean different in NuRadioReco and NuRadioMC. This is not at all obvious and can very easily trip someone up. Just to be clear: This particle_number only needs to be added to the hdf5 files. For the .nur files I think the other way is better. Wouldn't the run_number have to be added to the hdf5 files as well? Then it's just a matter of what to call that new entry and we might as well pick a less confusing name.

About splitting up the events: Why not implement a module for that? We could give the trigger modules the ability to add multiple triggers to the station if there are multiple pulses. Then this new module splits the event for each trigger and finally another module cuts the traces down to the correct size. That way it is transparent what is going on (it's not necessarily obvious that a trigger also cuts the trace) and we don't have to implement the event splitting and trace cutting for each trigger.
That would also make splitting up different raytracing solutions very easy: You just implement a kind of trigger module that looks at the simulated E-fields and sets triggers accordingly. Then you can use the same modules as with the regular triggers.

cg-laser, 2020-01-16:
> I don't think using the run_number and event_id that way is a good idea. For 2 reasons:
> 
> 1. In the real data, run_number and event_id will be set by the station, which of course has no idea if two events are related. We want to be able to stitch two events together later without changing run_number and event_id, so we need some other way to mark relations between events.
> 2. That would mean event_id and run_number can mean different things in NuRadioReco and NuRadioMC. This is not at all obvious and can very easily trip someone up. Just to be clear: This particle_number only needs to be added to the hdf5 files. For the .nur files I think the other way is better. Wouldn't the run_number have to be added to the hdf5 files as well? Then it's just a matter of what to call that new entry and we might as well pick a less confusing name.
> 
Good point. I agree. 

But there is a problem with your suggestion. If I understood it correctly, you want to increment the event id for cases where you need to split an event. E.g. you start simulating event number 1 which ends up having two very delayed ray tracing solutions, so you want to name the events 1 and 2. However, there is already an event number 2 in the hdf5 input file. So you can't dynamically increment event numbers. 

A way to solve that (and maybe you mean the same) is to introduce a layered approach in NuRadioMC that I called "event" and "sub_event_id". I think you refer to it as "particle_id" and "event_id". Not yet sure what makes more sense, as particle id is confusing. But your naming scheme is better because "event_id" in NuRadioMC will correspond to the "event_id" in NuRadioReco. But this would mean we need to change the definition of "event_id" in NuRadioMC.

> About splitting up the events: Why not implement a module for that? We could give the trigger modules the ability to add multiple triggers to the station if there are multiple pulses. Then this new [...]
Yes this would work, but is computationally (potentially much, depending on the separation) more expensive. The more elegant solution would be to split the events at an earlier stage and simulate the trigger on smaller chunks of the trace. But this is more difficult to implement, and maybe computing time is not that critical (?). 
> That would also make splitting up different raytracing solutions very easy: You just implement a kind of trigger module that looks at the simulated E-fields and sets triggers accordingly. Then you can use the same modules as with the regular triggers.

Yes would also work but you could have cases where both ray tracing solutions are close in time, so that you would have a trigger for both solutions although only one has a sufficiently high amplitude. However, this is just what it is, so also ok. 

We should also consider giving the hdf5 input files the possibility to already start with sub events. hmm but this will then lead to the same problem of not being able to dynamically increment event ids. I'm not yet sure about this point...

anelles, 2020-01-17:
I think the discussion is converging in a good direction. A couple of comments.
How events will be split will be very detector dependent. Some detectors have longer traces, other longer dead-times between events. So in simulation we will encounter two things: Simulate some generic detector that can keep all events and sub_events. And the true detector simulations in which we may loose signals. We somehow have to account for both. Currently, we implemented it in NuRadioMC to only account for the latter. 
So we may at some-point have different event_ids, so which is why I prefer something like 'associated' events where every event just has a field in which everything connected in stored. Also, the same shower may be seen in different stations, thus different event_ids also with different run_ids, depending on how the DAQ works. This not foreseen at all, right?

christophwelling, 2020-01-17:
> But there is a problem with your suggestion. If I understood it correctly, you want to increment the event id for cases where you need to split an event. E.g. you start simulating event number 1 [...]

I did not consider how the event_id would be set when we split an event up, that is indeed a problem. We had a discussion here and introducing a sub_event_id seems like the best option. So an event's "name" would be made of 3 parts: The run_number (which is always 0 in NuRadioMC), the event_id and the sub_event_id. So events can have the same event_id, as long as not all three are the same.

We found a problem with storing a list of associated events in the event, though. Let's say we have a double bang, so we split the event up into two events with the sub_event_ids 1 and 2. Then we reconstruct sub-event 1 and write it to file. Then sub-event 2 has a DnR pulse and is split by the trigger into events with the sub_event_ids 3 and 4 (or however we want the naming scheme to go). The problem now is that sub-event 1 has no way of knowing that it is associated with these new events. We can not change sub-event 1 because it is already written to disk.
The only solution I can think of is to introduce some event property called "event_group" or so. This would not be considered part of the "event name" (so it is allowed to change afterwards), but events would be associated by giving them the same "event_group".
I'm not sure if I like this idea very much myself, but it is the only one I could come up with. Maybe we should add an event group to the NuRadioReco event structure, above the event itself?

> We should also consider giving the hdf5 input files the possibility to already start with sub events. hmm but this will then lead to the same problem of not being able to dynamically increment event ids. I'm not yet sure about this point...

I don't quite understand what you mean here.

> How events will be split will be very detector dependent. Some detectors have longer traces, other longer dead-times between events. So in simulation we will encounter two things: Simulate some [...]

That would be solved by keeping the trigger modules, the module splitting up the events and the one cutting the traces separate. Any experiment would need to have an accurate trigger simulation that accounts for stuff like dead times anyway. Then the event splitter can use these and the trace cutter just needs to be set to the correct trace length.

> So we may at some-point have different event_ids, so which is why I prefer something like 'associated' events where every event just has a field in which everything connected in stored. Also, the [...]

I guess I was a bit unclear here. I was planning to store the associated events as a list of run_number/event_id pairs. So that would cover this situation, right?. But as I wrote, this approach has problems.

cg-laser, 2020-01-17:
> We found a problem with storing a list of associated events in the event, though. Let's say we have a double bang, so we split the event up into two events with the sub_event_ids 1 and 2. Then we [...]

why don't you just increment the event ids? So the second split would not result in 3 and 4 but 2 and 3. But I agree that it will be somewhat difficult to know which subevent belongs to what input condition. But this is anyway never clear. Because we merge all input signals (different subshowers, different ray tracing solutions) into one efield. And then calculate the trigger. So we never really know which part of the trace corresponds to which solution. You can only somewhat approximate it via the timing and amplitude of the simulated efield traces. 

And we also want it like that because this is a realistic simulation. The only other option would be to simulate each efield contribution separately, but then we would never take into account interference between different signals etc.

cg-laser, 2020-06-29:
implemented in #208

closed 2020-06-29 by cg-laser