nu-radio/NuRadioMC #234: Trigger count in Veff example

christophwelling, 2020-09-02:
The output in the effective volume example seems weird. It looks like the triggers are going down from 8 to 7 to 6

STATUS:NuRadioMC:processing 1252/1252 (8 triggered) = 100.0%, ETA 0s, time consumption: ray tracing = 69% (att. length 40%), askaryan = 1%, detector simulation = 26% reading input = 0%, calculating weights = 2%, distance cut 0%, unaccounted = 1% 
STATUS:NuRadioMC:start saving events
STATUS:NuRadioMC:fraction of triggered events = 7/1000 = 0.006 (sum of weights = 4.60)
STATUS:NuRadioMC:Veff = 0.624 km^3, Veffsr = 7.841 km^3 sr
STATUS:NuRadioMC:Timing of NuRadioReco modules 
efieldToVoltageConverter: 1s 49.4%
channelResampler: 0s 4.7%
channelBandPassFilter: 0s 14.2%
channelBandPassFilter: 0s 14.2%
triggerSimulator: 0s 9.7%
triggerSimulator: 0s 3.9%
triggerSimulator: 0s 3.9%

cg-laser, 2020-09-02:
this is because during the simulation every shower is counted. For the Veff calculation, duplicate triggeres are removed.

anelles, 2020-09-04:
@cg-laser should we then make the output more reasonable? It also confused me and I should have known.

fschlueter, 2025-05-02:
The logging has changed to not reporting the number of triggered events anymore during the processing:
```
STATUS - 2025-04-30 13:19:32,035 - NuRadioMC.simulation - Processing event group 49673/50000. ETA: 51s, time consumption: event builder (hdf5 -> nur) = 0%, weight calc. = 0%, ray tracing = 1%, ray tracing (time) = 0%, signal generation = 68%, propagation effects = 24%, detector response (sim) = 3%, group into events = 0%, detector response = 0%, trigger = 0%, unaccounted: 4%
STATUS - 2025-04-30 13:20:28,533 - NuRadioMC.HDF5OutputWriter - fraction of triggered events = 23/50000 (sum of weights = 22.30)
STATUS - 2025-04-30 13:20:28,533 - NuRadioMC.HDF5OutputWriter - Veff = 0.05697 km^3, Veffsr = 0.7159 km^3 sr
```

closed 2025-05-02 by fschlueter