nu-radio/NuRadioMC #239: Ray tracing solution appears twice

cg-laser, 2020-09-09:
In rare cases both the direct and reflected solution are returned for almost the same ray. Receive angles differ by a fraction of a degree
```
In [13]: sim_station.get_electric_fields()[0][efp.zenith]/units.deg
Out[13]: 37.990554573137445

In [14]: sim_station.get_electric_fields()[1][efp.zenith]/units.deg
Out[14]: 37.98455113752616

In [9]: sim_station.get_electric_fields()[0][efp.ray_path_type]
Out[9]: 'direct'

In [10]: sim_station.get_electric_fields()[1][efp.ray_path_type]
Out[10]: 'reflected'
```
This must be a bug as the direct solution should originate from below.

fschlueter, 2025-08-08:
@cg-laser can you provide a MWE for this? I.e. a geometry where this happens?

cg-laser, 2025-08-28:
as this report was 5 years ago, this might have well be solved by now, but I'm not sure. Unfortunately, I don't have an example at hand to verify it.