nu-radio/NuRadioMC #254: proposal volume extension

shallmann, 2020-12-11:
Volume extension by the 95% quantile of the tau-length seems in my opinion to be wrongly implemented:

https://github.com/nu-radio/NuRadioMC/blob/8d335c0dd0e6453ad70d9e1604eb0ea52a1442fe/NuRadioMC/EvtGen/generator.py#L438
https://github.com/nu-radio/NuRadioMC/blob/8d335c0dd0e6453ad70d9e1604eb0ea52a1442fe/NuRadioMC/EvtGen/generator.py#L497

If I choose a zenith range from 0 to pi, the max_horizontal_distance = np.abs(max(0, 0)*fiducial_zmin = 0
as a consequence the horizontal distance is not extended.

What is intended is to have a max_in_thetarange, i.e. some sampling like
```
dvals = np.linspace(attributes['thetamin'], attributes['thetamax'], 1000)
max_horizontal_dist = np.abs(max(np.abs(np.tan(dvals)))*volume['fiducial_zmin'])
```
would be a dirty fix.. any better ideas?

shallmann, 2020-12-11:
or @cg-laser maybe this extension for full zenith range is not wanted, since otherwise most simulated events will not be visible? ... in any case the implementation seems dodgy

(in addition: I haven't thought about if there is an energy range, where one could get visible secondary showers from mu but tau would decay with much shorter 95% quantile length)

anelles, 2021-01-05:
@shallmann did you compare whether this actually changes the results? I agree that your suggestion is a better implementation, but  I am wondering how big the difference actually is.

shallmann, 2021-01-07:
@anelles you mean an evaluation if the extension is relevant at all at trigger level?

I stumbled upon the previous implementation since it did not behave as promised. The effect... well it depends on the way events are simulated:
(a) Full cos(zenith) range: Old code will not extend the volume at all ...
(b) Simulation in small cos(zenith) bands: old code will give same (or round ~pi/2 at least similar) results ...

... if we stick with previous I would at least throw a warning in (a) that volume is not extended

cg-laser, 2021-01-07:
This was implemented for relatively narrow zenith angle bins. 
The problem of taking the largest value is that at 90deg, you would get an infinite extension. 

I agree, adding a warning at least would be good. It would be very inefficient to extend the volume and use a full zenith angle range as most of the events are nowhere near the detection volume. So this implementation makes sense for a reasonable/correct use of the module. I'm not sure how to really improve it. I think warning the user that he might be doing something stupid might be the best.

shallmann, 2021-12-09:
Fixed in pull request #335

closed 2021-12-09 by shallmann