nu-radio/NuRadioMC #258: Add linter to NuRadioMC

anelles, 2021-01-05:
Since we want to merge NuRadioReco over, we should also move the linter functionality over. Everyone needs to help though cleaning up the code. 
@christophwelling can you start in a PR and then we all help?
Or who volunteered to do this? I forgot ... sorry :/

christophwelling, 2021-01-06:
The Linter is already part of the new tests, but it only fails the test if there are serious issues.
We should do the code cleanup last, otherwise this will produce to many merge conflicts for the other PRs.

cg-laser, 2021-01-07:
agreed, and I'm also inclined to not be too strict on syntax. I think failing only on critical issues might be a good way. The linter still tells us other warnings. In any case, I would leave it like that for now.