nu-radio/NuRadioMC #262: Progress output seems off

christophwelling, 2021-01-14:
It looks like the output giving the progress for the simulation is wrong by roughly a factor 1/2. It finishes when its progress is just at 50-60%:

`
WARNING:SignalGen.ARZ:setting seed to 1235
WARNING:SignalGen.ARZ:loading shower library (/home/christoph/Software/NuRadioMC/NuRadioMC/SignalGen/ARZ/shower_library/library_v1.2.pkl) into memory
WARNING:NuRadioReco.antennapattern:loading antenna file vpol_4inch_center took 0 seconds
STATUS:NuRadioMC:processing event group 212/2000 and shower 137/2459 (1 showers triggered) = 5.6%, ETA 17m4s, time consumption: ray tracing = 16% (att. length 73%), askaryan = 81%, detector simulation = 3% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 319/2000 and shower 217/2459 (1 showers triggered) = 8.8%, ETA 21m9s, time consumption: ray tracing = 12% (att. length 74%), askaryan = 85%, detector simulation = 3% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 376/2000 and shower 267/2459 (1 showers triggered) = 10.9%, ETA 25m20s, time consumption: ray tracing = 11% (att. length 75%), askaryan = 85%, detector simulation = 4% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 459/2000 and shower 328/2459 (1 showers triggered) = 13.3%, ETA 27m0s, time consumption: ray tracing = 10% (att. length 76%), askaryan = 86%, detector simulation = 4% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 556/2000 and shower 408/2459 (6 showers triggered) = 16.6%, ETA 25m58s, time consumption: ray tracing = 9% (att. length 75%), askaryan = 86%, detector simulation = 5% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 665/2000 and shower 494/2459 (6 showers triggered) = 20.1%, ETA 24m47s, time consumption: ray tracing = 9% (att. length 75%), askaryan = 86%, detector simulation = 4% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 742/2000 and shower 558/2459 (6 showers triggered) = 22.7%, ETA 24m39s, time consumption: ray tracing = 9% (att. length 76%), askaryan = 87%, detector simulation = 4% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 835/2000 and shower 624/2459 (7 showers triggered) = 25.4%, ETA 24m17s, time consumption: ray tracing = 9% (att. length 76%), askaryan = 86%, detector simulation = 4% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 1016/2000 and shower 753/2459 (8 showers triggered) = 30.6%, ETA 20m59s, time consumption: ray tracing = 9% (att. length 76%), askaryan = 86%, detector simulation = 5% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 1139/2000 and shower 854/2459 (9 showers triggered) = 34.7%, ETA 19m22s, time consumption: ray tracing = 9% (att. length 76%), askaryan = 87%, detector simulation = 4% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 1271/2000 and shower 948/2459 (10 showers triggered) = 38.6%, ETA 18m8s, time consumption: ray tracing = 9% (att. length 76%), askaryan = 87%, detector simulation = 5% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 1380/2000 and shower 1035/2459 (13 showers triggered) = 42.1%, ETA 17m2s, time consumption: ray tracing = 9% (att. length 76%), askaryan = 87%, detector simulation = 4% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 1453/2000 and shower 1088/2459 (16 showers triggered) = 44.2%, ETA 16m54s, time consumption: ray tracing = 8% (att. length 76%), askaryan = 87%, detector simulation = 5% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 1608/2000 and shower 1211/2459 (18 showers triggered) = 49.2%, ETA 14m53s, time consumption: ray tracing = 8% (att. length 76%), askaryan = 87%, detector simulation = 5% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 1695/2000 and shower 1277/2459 (20 showers triggered) = 51.9%, ETA 14m21s, time consumption: ray tracing = 8% (att. length 76%), askaryan = 87%, detector simulation = 5% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 1855/2000 and shower 1407/2459 (20 showers triggered) = 57.2%, ETA 12m21s, time consumption: ray tracing = 8% (att. length 76%), askaryan = 87%, detector simulation = 5% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:processing event group 1955/2000 and shower 1488/2459 (20 showers triggered) = 60.5%, ETA 11m27s, time consumption: ray tracing = 8% (att. length 76%), askaryan = 87%, detector simulation = 5% reading input = 0%, calculating weights = 0%, distance cut 0%, unaccounted = 0% 
STATUS:NuRadioMC:start saving events
STATUS:NuRadioMC:fraction of triggered events = 16/2000 = 0.007 (sum of weights = 11.84)
STATUS:NuRadioMC:Veff = 0.8034 km^3, Veffsr = 10.1 km^3 sr
STATUS:NuRadioMC:Timing of NuRadioReco modules 
efieldToVoltageConverter: 6s 14.5%
channelResampler: 0s 0.0%
hardwareResponseIncorporator: 35s 74.4%
triggerSimulator: 2s 4.4%
channelSignalReconstructor: 3s 6.6%

STATUS:NuRadioMC:2459 events processed in 17m45s = 433.16ms/event (0.0% input, 8.3% ray tracing, 86.8% askaryan, 4.8% detector simulation, 0.0% output, 0.1% weights calculation)
`

cg-laser, 2021-01-14:
I think this is because two stations are simulated where one of the station never participates? I need to look into this. An easy fix would be to give the percentage and the time estimate only based on the event group and not the shower count.

christophwelling, 2021-01-14:
This one was done with only 1 station.

cg-laser, 2021-01-14:
what version did you use?

christophwelling, 2021-01-14:
The current master

fschlueter, 2025-05-02:
Since the logging has changed I will close this.

closed 2025-05-02 by fschlueter