nu-radio/NuRadioMC #265: Can't handle non-sequential channel IDs

christophwelling, 2021-01-26:
simulation.py  throws an error if the channel IDs in the detector description are not sequential.
This is because in line 261 it is assumed that the channel IDs are 
`range(self._det.get_number_of_channels(self._station_id)).`
I don't know if changing this would cause problems somewhere else in the code

cg-laser, 2021-01-26:
Didn't I change that the detector returns the channel ids always in increasing order? 
Oh but in this case some channel_ids are missing, e.g. [0, 5, 9]?

I think we should be careful changing that. It might be that several modules assume consecutive channel ids, although they shouldn't. The interfaces are such that this is not a hard requirement.

fschlueter, 2025-05-02:
I think this is implemented since NuRadio 3.0 fft repopen if I am wrong

closed 2025-05-02 by fschlueter