nu-radio/NuRadioMC #286: number of solutions for southpole_2015 model unexpectadly zero

jakob2508, 2021-04-23:
I'm using the analytic ray tracer for the southpole 2015 ice model to find the trajectories for many different starting positions. When scanning over these positions I expect to cross over to the shadow zone, however at some positions within the non-shadow zone I get zero solutions from `get_number_of_solutions()`. Attached you find the minimal example code for a specific position that fails. The image shows the issue on a larger scale. Purple dots indicate `get_number_of_solutions()=2`, while the blanks represent `get_number_of_solutions()!=2`. The red circled region highlight the blanks.

```
from NuRadioMC.SignalProp import propagation
from NuRadioMC.utilities import medium
from NuRadioReco.utilities import units
import numpy as np

prop = propagation.get_propagation_module('analytic')
ref_index_model = 'southpole_2015'
ice = medium.get_ice_model(ref_index_model)
attenuation_model = 'SP1'

start = np.array([-90, 0, -10])*units.m
end = np.array([0, 0,-15])*units.m

rays = prop(ice, attenuation_model,
                    n_frequencies_integration=25,
                    n_reflections=0)
rays.set_start_and_end_point(start, end)
rays.find_solutions()
number_of_solutions = rays.get_number_of_solutions()
number_of_raytracing_solutions = rays.get_number_of_solutions()
```

[image]

I don't expect the transition between the non-shadow and shadow zone to have any of these sudden blank spots.

I found that though `get_number_of_solutions()=0` at these blank spots `get_number_of_raytracing_solutions()=2`. However, `get_path()` has no entry. What's the difference between the two ?

cg-laser, 2021-04-26:
`get_number_of_raytracing_solutions` returns the maximum possible number of RT solutions. I suggest to change this to `get_max_number_of_possible_raytracing_solutions` or something similar that is more descriptive. @boboeyen can you do this in the radio propa PR? 

@boboeyen can you run radiopropa on the two points that Jakob provided to see if you get a solution?

cg-laser, 2021-04-26:
I checked that I only get the problem when using the Cpp implementation. The python implementation gives 2 solutions

cg-laser, 2021-04-26:
I activated some debug options
```
DEBUG:ray_tracing:swap = True
DEBUG:ray_tracing:X1 = [  0.   0. -15.], X2 = [-90.   0. -10.]
DEBUG:ray_tracing:dphi = -180.0
DEBUG:ray_tracing:X2 - X1 = [-90.   0.   5.], X1r = [  0.   0. -15.], X2r = [ 9.00000000e+01  1.10218212e-14 -1.00000000e+01]
DEBUG:ray_tracing:2D points [  0. -15.] [ 90. -10.]
Got to iter 1 val is 3533.55
Got to iter 2 val is 1904.51
Got to iter 3 val is 2859.95
Got to iter 4 val is 4922.34
Got to iter 5 val is 3.38948e+06
Got to iter 6 val is 4932.59
Got to iter 7 val is 4942.84
Got to iter 8 val is 640534
Got to iter 9 val is 4980.98
Got to iter 10 val is 5019.22
Got to iter 11 val is 715093
Got to iter 12 val is 5054.27
Got to iter 13 val is 5089.4
Got to iter 14 val is 811370
Got to iter 15 val is 5121.17
Got to iter 16 val is 5152.99
Got to iter 17 val is 910046
Got to iter 18 val is 5182.04
Got to iter 19 val is 5211.12
Got to iter 20 val is 1.01053e+06
Got to iter 21 val is 5237.88
Got to iter 22 val is 5264.66
Got to iter 23 val is 1.11285e+06
Got to iter 24 val is 5289.47
Got to iter 25 val is 5314.29
Got to iter 26 val is 1.21705e+06
Got to iter 27 val is 5337.41
Got to iter 28 val is 5360.53
Got to iter 29 val is 1.32316e+06
Got to iter 30 val is 5382.17
Got to iter 31 val is 5403.81
Got to iter 32 val is 1.43118e+06
Got to iter 33 val is 5424.14
Got to iter 34 val is 5444.48
Got to iter 35 val is 1.54113e+06
Got to iter 36 val is 5463.65
Got to iter 37 val is 5482.82
Got to iter 38 val is 1.65301e+06
Got to iter 39 val is 5500.94
Got to iter 40 val is 5519.07
Got to iter 41 val is 1.76681e+06
Got to iter 42 val is 5536.25
Got to iter 43 val is 5553.44
Got to iter 44 val is 1.88254e+06
Got to iter 45 val is 5569.77
Got to iter 46 val is 5586.1
Got to iter 47 val is 2.00019e+06
Got to iter 48 val is 5601.65
Got to iter 49 val is 5617.2
Got to iter 50 val is 2.11977e+06
Got to iter 51 val is 5632.04
Got to iter 52 val is 5646.88
Got to iter 53 val is 2.24127e+06
Got to iter 54 val is 5661.06
Got to iter 55 val is 5675.25
Got to iter 56 val is 2.36469e+06
Got to iter 57 val is 5688.83
Got to iter 58 val is 5702.41
Got to iter 59 val is 2.49001e+06
Got to iter 60 val is 5715.43
Got to iter 61 val is 5728.45
Got to iter 62 val is 2.61725e+06
Got to iter 63 val is 5740.95
[... 115 lines cut ...]
Got to iter 179 val is 9.03888e+06
Got to iter 180 val is 6308.94
Got to iter 181 val is 6313.33
Got to iter 182 val is 9.24041e+06
Got to iter 183 val is 6317.64
Got to iter 184 val is 6321.94
Got to iter 185 val is 9.44377e+06
Got to iter 186 val is 6326.17
Got to iter 187 val is 6330.39
Got to iter 188 val is 9.64894e+06
Got to iter 189 val is 6334.54
Got to iter 190 val is 6338.68
Got to iter 191 val is 9.85593e+06
Got to iter 192 val is 6342.76
Got to iter 193 val is 6346.82
Got to iter 194 val is 1.00647e+07
Got to iter 195 val is 6350.82
Got to iter 196 val is 6354.82
Got to iter 197 val is 1.02754e+07
Got to iter 198 val is 6358.74
Got to iter 199 val is 6362.66
Got to iter 200 val is 1.04878e+07
NOT converged on root 1! Iteration 200
```
For some reason the Cpp minimizer is not converging. @clark2668 you implemented this originally. Any ideas?

clark2668, 2021-04-29:
Sorry it took me so long to look at this. I'm taking a look at it. Short answer: the initial conditions of the minimization don't seem to allow GSL to find solutions for some reason. When I tweak the initial conditions, e.g. https://github.com/nu-radio/NuRadioMC/commit/eb52bc226ac551e5ad61be39f0d704069ebf2667, I can get this specific case to go, but then tests fail: https://github.com/nu-radio/NuRadioMC/runs/2470634943?check_suite_focus=true. So I'm not sure what's up yet, might have to debug more.

clark2668, 2021-04-29:
Okay, I think I've got a working solution in the `fix_cpp_raysolver` branch. Can you give it a try?

PR #288 (2021-04-30): Fix cpp raysolver
this PR fixes #286 where the Cpp implementation does not find a solution. Thanks for @clark2668 for fixing it. 
The solution is to try several initial values. For most cases this should not increase the computing time as the minumum should be found by the first trial.
closed 2021-04-30 by cg-laser