nu-radio/NuRadioMC #320: pyproject.toml is broken, fix or just use setuptools?

cozzyd, 2021-10-05:
With a well-formed pyproject.toml, `pip install .` should install all requirements, generate the wheels for NuRadioReco and NuRadioMC and install them somewhere reasonable (typically whatever virtualenv you're running in, unless you're brave enough to run this outside a virtualenv...). 

This fails, firstly because the pyproject.toml declares a dpeendency on pickle, which is a built-in module that pip doesn't know how to install, but this is easy to fix. After fixing this, `pip install .` will download the requirements for and install NuRadioMC, so that part is good. 

But it doesn't install NuRadioReco,and I don't know how to fix that since I don't know if flit supports multiple modules. Trying to hack this by just changing the module to NuRadioReco in pyproject.toml also fails because it refuses to install a package with no docstring. After adding a docstring, this works and you have a usable, importable NuRadioReco and NuRadioMC without the indignity of simply adding the git checkout to the `$PYTHONPATH.`

It would be good if the project metadata can be fixed to allow local installation via `pip install .` as that would greatly simplify the installation process for many people. I would do it, but I don't know how to use flit, and googling how to specify multiple modules doesn't come up with a quick answer. It may be necessary to have a pyproject.toml for each subproject instead, or to revert back to good old requirements.txt/setup.py  that I vaguely know how to use (and could probably handle compiling any compiled code... not sure if flit knows how to do that or not). Or maybe just add a requirements.txt so people can `pip install -r requirements.text` and ` export PYTHONPATH=$PYTHONPATH:$CWD ` and that's not too bad either.

anelles, 2021-12-08:
@cozzyd do you want to check whether this new approach is better and would work for you?

cozzyd, 2021-12-09:
I can confirm that pip install . now at least appears to work on this branch! Though I haven't tested if all the dependencies are actually installed. 

I see there is also an install-dev.py script but it's not clear to me exactly what it does (but I imagine it will be documented later).

closed 2021-12-20 by sjoerd-bouma