nu-radio/NuRadioMC #351: Install dependencies in run_tests.yaml overwrites (?) code actually in commit

sjoerd-bouma, 2021-12-06:
The automatic tests in the github workflow run_tests.yaml overwrite code in the NuRadioReco module with code from a zip archive:
```
wget https://github.com/nu-radio/NuRadioReco/archive/refactor-pa.zip -O /tmp/NuRadioReco.zip
unzip /tmp/NuRadioReco.zip
mv NuRadioReco-refactor-pa/NuRadioReco $PWD/NuRadioReco
```

This means that instead of testing the code in the pushed commit, it will actually run the tests on code in this archive (which will trivially complete).

I ran into this when testing the documentation. If there is some very good reason to test this archive, I suggest we make a separate test for it. Otherwise, this should just be removed from `run_tests.yaml` ASAP.

sjoerd-bouma, 2021-12-06:
@cg-laser Looks like github blames you for this... Please let me know if there's some reason to keep this, otherwise I will make a PR to remove it.

anelles, 2021-12-08:
This looks like it was a temporary test, when we were moving from the old repo to the new repo, respectively when we wanted to test something with a something not in the master. Please make a PR to fix this, yes!

cg-laser, 2021-12-08:
thanks for catching this and providing a fix!

PR #353 (2021-12-09): Fix NuRadioReco being overwritten in tests
The unit tests (run_test.yaml) workflow used to overwrite the NuRadioReco module with an older version from a zipped archive, and thus would potentially not test the committed code properly. I've removed this line (as well as some commented out lines) from run_tests.yaml.

Fixes #351
closed 2021-12-09 by sjoerd-bouma