nu-radio/NuRadioMC #352: Potential issue in Hilbert envelope?

shallmann, 2021-12-08:
I was trying to use the get_hilbert_envelope function in the `BaseTrace` class, and didn't return what I expected (i expected `np.array(np.abs(h))`. Instead, only the first three items are returned, is that intended? Why do we want that?

https://github.com/nu-radio/NuRadioMC/blob/572d4f6382db4928b75581dda117c3f9e90b013d/NuRadioReco/framework/base_trace.py#L129

anelles, 2021-12-08:
This looks like it was made to return the hilbert envelope for three components of the electric field. So the main axis seems to be wrong for the type of array that you are using this for. This shouldn't happen.

anelles, 2021-12-08:
@cg-laser actually noticed this already in #304 , so this needs fixing.

shallmann, 2021-12-09:
Fixed

closed 2021-12-09 by shallmann