nu-radio/NuRadioMC #364: Need to improve proposal directories to not fail when tau & mu are produced at the same time

shallmann, 2022-01-08:
When tau and mu are being produced at the same time (from different jobs using the same dir), an error is thrown:
```
Traceback (most recent call last):
  File "generate_step1_file.py", line 104, in <module>
    proposal_kwargs={'low_nu': minEproposal*units.eV, 'min_energy_loss_nu': minEproposal*units.eV})
  File "/afs/ifh.de/group/radio/scratch/shallman/software/NuRadioMC/NuRadioMC/EvtGen/generator.py", line 1326, in generate_eventlist_cylinder
    **proposal_kwargs)
  File "/afs/ifh.de/group/radio/scratch/shallman/software/NuRadioMC/NuRadioMC/EvtGen/NuRadioProposal.py", line 615, in get_secondaries_array
    propagation_length, low=low)
  File "/afs/ifh.de/group/radio/scratch/shallman/software/NuRadioMC/NuRadioMC/EvtGen/NuRadioProposal.py", line 495, in __propagate_particle
    secondaries = self.__get_propagator(lepton_code).propagate(initial_condition,
  File "/afs/ifh.de/group/radio/scratch/shallman/software/NuRadioMC/NuRadioMC/EvtGen/NuRadioProposal.py", line 357, in __get_propagator
    self.__propagators[particle_code] = pp.Propagator(particle_def=mu_def, config_file=config_file_full_path)
IndexError: vector::_M_range_check
```
I *think* this should at some point be looked at and prevented (I would have expected that this does not occur since the proposal files have some unique extensions)

fschlueter, 2025-08-08:
@shallmann is that still relevant? Can you provide some more information? I.e. what directories? Is proposal storing particle information in files on the harddrive while running?

shallmann, 2025-08-08:
This had been fixed if I am not mistaken