nu-radio/NuRadioMC #367: NuRadioMC dependencies not up to date?

sjoerd-bouma, 2022-01-12:
Currently, NuRadioMC dependencies are listed in 3 different places - the [wiki](https://github.com/nu-radio/NuRadioMC/wiki#install-dependencies), the [documentation](https://nu-radio.github.io/NuRadioMC/Introduction/pages/installation.html), and the `pyproject.toml` file. 

- The `pyproject.toml` is probably the most relevant one, as it's used to determine dependencies for e.g. online tests and the Pypi releases. It should be mostly up to date... caveats below
- I have updated the wiki to reflect `pyproject.toml`. However, if we're serious about moving the documentation to sphinx, we should probably remove the wiki at some point... (see also #277)
- I will update the documentation soon; almost all of the wiki pages have now been ported over, just need to make another PR.

**Unnecessary dependencies (?)**
- `awkward` - I don't *think* this is actually used anywhere?

**Missing dependencies**
- `scikit-image` - used for debug plots in the `neutrinoVertexReconstructor` module. Should probably be a core dependency?

cg-laser, 2022-01-12:
Thanks for working on this. I agree, we should remove the wiki soon, or at least the pages that are already in the documentation and link to the documentation from the wiki. 

Isn't awkward used by uproot? 

Ii'm not sure about the debug plot dependencies. This feels rather like an optional dependnecy

sjoerd-bouma, 2022-03-31:
For completeness - the `scikit-image` dependency is only used in the `neutrino2DVertexReconstructor` module, which will probably soon be deprecated, so I'm not going to add it. `awkward` and `uproot` are kept as core dependencies.