nu-radio/NuRadioMC #369: re-writing a .nur file with missing sim traces creates broken .nur file

sjoerd-bouma, 2022-01-12:
Reading a `.nur` file with 'missing' `sim_channel_traces` (e.g., because they were not saved in the simulation) using the `eventReader` module, and then writing the read event using the `eventWriter` module results in an error when reading the resulting `.nur` file:

```
(...)
~/Python/software/nu-radio/NuRadioMC/NuRadioReco/framework/electric_field.py in deserialize(self, data_pkl)
    141         data = pickle.loads(data_pkl)
    142         if(data['base_trace'] is not None):
--> 143             NuRadioReco.framework.base_trace.BaseTrace.deserialize(self, data['base_trace'])
    144         if 'position' in data:  # for backward compatibility
    145             self._position = data['position']

~/Python/software/nu-radio/NuRadioMC/NuRadioReco/framework/base_trace.py in deserialize(self, data_pkl)
    193     def deserialize(self, data_pkl):
    194         data = pickle.loads(data_pkl)
--> 195         self.set_trace(data['time_trace'], data['sampling_rate'])
    196         if('trace_start_time' in data.keys()):
    197             self.set_trace_start_time(data['trace_start_time'])

~/Python/software/nu-radio/NuRadioMC/NuRadioReco/framework/base_trace.py in set_trace(self, trace, sampling_rate)
     78         """
     79         if trace is not None:
---> 80             if trace.shape[trace.ndim - 1] % 2 != 0:
     81                 raise ValueError('Attempted to set trace with an uneven number ({}) of samples. Only traces with an even number of samples are allowed.'.format(trace.shape[trace.ndim - 1]))
     82         self.__time_domain_up_to_date = True

IndexError: tuple index out of range
```
The trace is something like `np.array(None)`, i.e. returns `True` on `trace is not None`, but has shape `()`, leading to the above error.
A workaround is to replace the check `if trace is not None` by `if np.any(trace != None)` (suggested by @shallmann), though I'm not sure if it's a fix or a workaround for some bug in `eventWriter` that we're missing.

christophwelling, 2022-01-26:
The error is coming from the electric field, so it looks to me like that is where the empty trace is. What makes you think it is the sim_channel?

The best solution would be to prevent this weird object from showing up in the first place (though that would not fix reading files that already have this issue). Can you give a minimum working example to create such a file, so I can track down where exactly the problem is?

sjoerd-bouma, 2022-09-22:
Apologies for ignoring this for so long... I wasn't able to reproduce this last time I looked at it. This time I did manage to reproduce the `np.array(None)` objects, though... no error was raised? In any case, to reproduce this, take any event containing sim_channel traces (or ElectricField traces, ...), then save the same event using the `eventWriter` but don't save the traces.
```
from NuRadioReco.modules.io import eventReader, eventWriter

eventreader = eventReader.eventReader()
eventwriter = eventWriter.eventWriter()

eventreader.begin("test_nur.nur")
eventwriter.begin("output_nur.nur")

mode = dict(Channels=True, SimChannels=False, ElectricFields=False, SimElectricFields=False)

for evt in eventreader.run():
    eventwriter.run(evt, mode=mode)

eventreader.end()
eventwriter.end()
```
Rather than creating `base_trace` objects with  trace `None`, the trace will be `np.array(None)`

Will look into this tomorrow / next week if I find the time. If I still can't reproduce the error I think this issue can be closed.