nu-radio/NuRadioMC #380: Need simple example for how to read NUR files

clark2668, 2022-02-03:
It would be good if there was a simple example for how to read and navigate the .nur file format. For example, how to do things like loop over events, get receive angles, focusing factors, etc. Many of the existing examples are more advanced than this.

If it exists somewhere, I cannot find it, so clearing up how to find it would also be good!

sjoerd-bouma, 2022-02-04:
I guess [this](https://github.com/nu-radio/NuRadioMC/blob/develop/NuRadioReco/examples/SimpleNeutrinoReconstruction.py) is _too_ simple? There is also some information on the .nur file format and how to use it in the documentation [here](https://nu-radio.github.io/NuRadioMC/NuRadioReco/pages/event_structure.html). More functions and parameters for the `event`, `station` `channel` and `shower` classes are documented in the code (also in the online documentation).

I can see the need for a clearer named and more extensive example, though. I also think that the `.nur` file does not contain some of the things you're interested in (receive angles, focusing factors) - they can be found in the `.hdf5` file, but to get them from the `.nur` file I think you might have to resimulate the rays using the ray tracing module (at least I don't see these attributes in the `sim_channel`, where I would have expected to find them).

cg-laser, 2022-02-04:
the receive direction should be the `zenith` `azimuth` parameter of the simulated efields. But it might be that the focussing factor is missing in the nur files (I haven't checked). We are still in the process of getting all information of the hdf5 files also into the nur files. It seems that we might have overlooked the focussing factor.

clark2668, 2022-02-04:
I actually would have said it was still not simple enough :) So like, how would I know I need to check the simulated efield to get zenith and azimuth? How would I even discover the existence of the simulated efields in the file? Just some examples demonstrating how to discover and then access things would be really helpful.

Perhaps I didn't just read the docs thoroughly enough. But I suspect many people (guilty as charged) will be interested in "just" opening up a file and trying to see what's inside, and quickly being able to pull out some quantities.

christophwelling, 2022-02-16:
There is a documentation page on how to access data: https://nu-radio.github.io/NuRadioMC/NuRadioReco/pages/event_structure.html
It is a little outdated, but I am working on updating it.

fschlueter, 2025-08-08:
@clark2668, in the meantime we have updated the RNO-G examples in NuRadioReco. Are you happy with them? If not please provide explicit suggestions to improve/simplify them. Have you read the documentation page regarding the event structure? Do you have suggestions on how to improve it?

fschlueter, 2025-08-08:
I wonder a bit what we can do to make as easy as possible. Of course we can always improve discoverability of example scripts and documentation but also just to a certain point. I am getting the feeling you wish to have a `dataio-shovel` like tool such that students can graphically inspect nur files. I do not think we have to capacity to provide that.

fschlueter, 2025-10-15:
There have been many advances in the last 3 years in terms of documentation. Please reopen if you feel this is still a point to improve on

closed 2025-10-15 by fschlueter