nu-radio/NuRadioMC #382: Propagation of signal in two different ways for the emitter

boboeyen, 2022-02-08:
The following code in `simulation.py` calculates the electric field traces for an emitter. 
```
eTheta = VEL['theta'] * (-1j) * voltage_spectrum_emitter * frequencies * n_index / (c) * np.exp(-1j * k * R)
ePhi = VEL['phi'] * (-1j) * voltage_spectrum_emitter * frequencies * n_index / (c) * np.exp(-1j * k * R)
```
To my understanding, the complex exponential adds a phase which represents a time shift because of the propagation along path R. However, this time shift is also applied by the raytracer itself. So this results in a double time shift. Also, because the traces are not infinite, I have the feeling this exponential is a cyclic procedure. So when the time shoft due to the phase is larger then the trace, it is shifted to the front of the trace, resulting in weird timeshifts. 

I propose to just remove this exponential and then it should be fixed. What do you think @pgiri4 ?

boboeyen, 2022-02-08:
[image]
[image]

cg-laser, 2022-02-08:
@boboeyen what are the two plots showing?

pgiri4, 2022-02-08:
Thanks Bob for raising this issue. @ikrav and I were first developing the model without phase ( as you can see Ilya's slides that I sent to you). As far I remember, I introduced this factor after few people suggested me to consider the phase too. I can see reflection phase applied in raytracing code. I think your proposal to erase this factor is right. Let's see what @cg-laser and @ikrav has to say on this.

boboeyen, 2022-02-09:
@cg-laser both plots show the timing of the electric field traces in different channels when using an emitter. The top plot is with the phase factor in the code, and as you can see, the pulse do not align with the ray tracing timing. They are shifted and wrapped around by the phase factor. When I remove this factor you get the bottom plot in which everything seems fine

closed 2022-04-13 by boboeyen