nu-radio/NuRadioMC #385: Channel trace length

shallmann, 2022-02-25:
In the TDR sims, according to the detector description, the channels have
```
 "adc_n_samples": 2048
```
When I look at the `.nur` output the `len(channel.get_trace()))` returns `2046`. I think this is unexpected and should be looked at?

sjoerd-bouma, 2023-12-17:
This is most likely caused by running the triggerTimeAdjuster when the channel sampling rate is different from the target detector sampling rate (which can happen as the simulation is usually run at a higher internal sampling rate). The solution would be to downsample to the target sampling rate before running the triggerTimeAdjuster. 

#526 does a slightly better job of trying to ensure the correct number of samples in either case, and also now raises a warning if the sampling rate does not match the detector sampling rate.

fschlueter, 2025-05-02:
The linked module is now deprecated and other things changed to. I will close this issue. Feel free to reopen

closed 2025-05-02 by fschlueter