nu-radio/NuRadioMC #391: Wrong showers written to `station_xxx` substructure in .hdf5 output

shallmann, 2022-03-29:
**Describe the issue**
In the hdf5 output the wrong `shower_id`s are written to the nested tables for each station. This leads to max receive vectors being all nan for some showers

**Expected behavior**
In the `station_xxx` sub-tables the shower ids should match the ones triggered in the main table

**Additional context**
I am attaching a notebook that demonstrates the issue

**To-dos**

- [ ] Find why wrong showers are written and fix

- [ ] Add test checking not all receive vectors are nan for each showers, i.e something like
```
receive_vectors = np.array(f['station_1001']['receive_vectors'])
# for each shower, see if ANY receive vector (in any antenna) is not NaN... this should never happen
[np.isnan(receive_vectors[i,:,:]).all() for i in range(np.shape(receive_vectors)[0])]
```

shallmann, 2022-03-29:
[hdf5_shower_numbering.tar.gz](https://github.com/nu-radio/NuRadioMC/files/8371428/hdf5_shower_numbering.tar.gz)

shallmann, 2022-09-16:
@lpyras maybe you can check if this is an issue and tackle this issue when simulating atm muons?

cg-laser, 2022-09-19:
Hej Steffen, I think Nans can occur under certain conditions. The fixed dimensionality of the hdf5 output arrays forces it to contain always all antennas and ray tracing solutions. It might be that certain antennas don't have any RT solutions. Then their values are set to Nan. However that should not happen is that all receive vectors of a shower are Nan. 
But you also say that the shower ids are wrong? The shower ids in the station_xxx tables should be a subset of the shower_ids on global detector level.

fschlueter, 2023-02-23:
This can be close right?

closed 2023-02-23 by fschlueter