nu-radio/NuRadioMC #397: Wrong group delay antenna model RNO VPol

boboeyen, 2022-04-13:
The group delay of the RNO Vpol is negative (~ -10ns) which is unphysical and should be fixed. Below you can see the plots picturing the behaviour of different antenna models and one can clearly see that only the RNO Vpol has this issue.

boboeyen, 2022-04-13:
`
[image]
[image]
`

fschlueter, 2023-02-23:
Is this still relevant? At last it should be fixed with #492 correct?

cg-laser, 2023-02-23:
I don't think this was ever fixed. It would require a resimulation of the Vpol response with XFDTD but I don't think Dan ever did this, or at least it didn't end up as an updated antenna model in NuRadioMC.

fschlueter, 2023-02-24:
Ah probably I confused the group delay of the readout chain with that of the antenna itself. My bad...

MartinRavn, 2026-10-09:
I have plotted the group delays of many of the VPol and HPol models in NuRadioMC. The different models have very inconsistent group delays, probably because of inconsistent observer positions in the XF simulations. We are currently using `RNOG_vpol_v3_5inch_center_n1.74` and `RNOG_hpol_v4_8inch_center_n1.74`, which are very inconsistent. I have plotted Bryan's new simulations in black, which seem to be consistent with the relative group delays measured in the lab.

[image]

[image]

MartinRavn, 2026-10-09:
This is how I calculated the group delays
```
antenna_pattern = AntennaPattern(antnna_model)
antenna_response = antenna_pattern.get_antenna_response_vectorized(frequencies, theta, phi, 0, 0, 90*units.deg, 90*units.deg)
antenna_response_phase = np.unwrap(np.angle(antenna_response[component]))
antenna_response_phase_slope = np.diff(antenna_response_phase) / np.diff(frequencies)
group_delay = -antenna_response_phase_slope / (2 * np.pi)
```

MartinRavn, 2026-10-09:
And here are the LPDA models, which look better 

[image]

MartinRavn, 2026-10-09:
I think we should add the new simulations by @blhendricks to the antenna models and add warnings to the documentation. We can maybe also add a warning to the code if an old model is chosen.