nu-radio/NuRadioMC #411: pip install doesn't work from environment without scipy

jlazar17, 2022-06-22:
**Describe the issue**
When installing to a clean conda environment running Python3.7 without a scipy installation, import does not work producing the following error: 
```
~/anaconda3/lib/python3.6/site-packages/NuRadioReco/detector/filterresponse.py in <module>
      1 import numpy as np
      2 from NuRadioReco.utilities import units
----> 3 from scipy import interpolate as intp
      4 import os
      5 import glob

ImportError: cannot import name 'interpolate'
``` 
It seems that somehow pip is not installing the necessary modules from scipy

Full error message:
```
ImportError                               Traceback (most recent call last)
<ipython-input-2-d95f828c13c1> in <module>
----> 1 import NuRadioReco.modules.io.eventReader

~/anaconda3/lib/python3.6/site-packages/NuRadioReco/modules/io/eventReader.py in <module>
      1 from __future__ import absolute_import, division, print_function, unicode_literals
----> 2 from NuRadioReco.modules.base.module import register_run
      3 from NuRadioReco.modules.io import NuRadioRecoio
      4 import logging
      5 logger = logging.getLogger('eventReader')

~/anaconda3/lib/python3.6/site-packages/NuRadioReco/modules/base/module.py in <module>
      1 from functools import wraps
      2 from timeit import default_timer as timer
----> 3 import NuRadioReco.framework.event
      4 import NuRadioReco.framework.base_station
      5 import logging

~/anaconda3/lib/python3.6/site-packages/NuRadioReco/framework/event.py in <module>
      1 from __future__ import absolute_import, division, print_function
      2 import pickle
----> 3 import NuRadioReco.framework.station
      4 import NuRadioReco.framework.radio_shower
      5 import NuRadioReco.framework.hybrid_information

~/anaconda3/lib/python3.6/site-packages/NuRadioReco/framework/station.py in <module>
      1 from __future__ import absolute_import, division, print_function
----> 2 import NuRadioReco.framework.base_station
      3 import NuRadioReco.framework.sim_station
      4 import NuRadioReco.framework.channel
      5 from six import iteritems

~/anaconda3/lib/python3.6/site-packages/NuRadioReco/framework/base_station.py in <module>
      1 from __future__ import absolute_import, division, print_function, unicode_literals
----> 2 import NuRadioReco.framework.base_trace
      3 import NuRadioReco.framework.trigger
      4 import NuRadioReco.framework.electric_field
      5 import NuRadioReco.framework.parameters as parameters

~/anaconda3/lib/python3.6/site-packages/NuRadioReco/framework/base_trace.py in <module>
      5 import decimal
      6 import numbers
----> 7 from NuRadioReco.utilities import fft, bandpass_filter
      8 import scipy.signal
      9 import copy

~/anaconda3/lib/python3.6/site-packages/NuRadioReco/utilities/bandpass_filter.py in <module>
      1 import numpy as np
      2 import scipy.signal
----> 3 from NuRadioReco.detector import filterresponse
      4
      5

~/anaconda3/lib/python3.6/site-packages/NuRadioReco/detector/filterresponse.py in <module>
      1 import numpy as np
      2 from NuRadioReco.utilities import units
----> 3 from scipy import interpolate as intp
      4 import os
      5 import glob

ImportError: cannot import name 'interpolate'
```

sjoerd-bouma, 2022-07-12:
Hi, sorry for the late response. How did you install NuRadioMC? (i.e. `pip install NuRadioMC` or did you get the development version from github?). And are you sure you're using a clean conda environment on Python 3.7, which was active both during installation and when running the script? Your error message seems to imply the packages used are installed under `/anaconda3/lib/python3.6`

jlazar17, 2022-08-02:
I just went to recreate this and realized I had not installed ipython in the conda environment. Thus, ipython and python were pointing to different executables / different pip options. This was a misunderstanding on my part 🙃

closed 2022-08-02 by jlazar17