nu-radio/NuRadioMC #416: Simulation still requires default_station to be passed

christophwelling, 2022-07-21:
The simulation.py code still requires the user to pass a default_station ID in order to use a GenericDetector. Since this way to use the GenericDetector is deprecated, there should be another way to specify that a GenericDetector should be used.

cg-laser, 2022-07-25:
good point. What is your suggestion? Do we ever need to pass a default station id? 
Can the detector class determine automatically if the input file is a generic detector? That would be the preferred option then.

christophwelling, 2022-07-25:
In principle, we could check if there are reference stations and reference channels in the detector description and then use the right detector class automatically. That could break some things though, for people who still specify these manually.

cg-laser, 2022-07-25:
to not break backward compatibility right away, we could just ignore the additional `reference_station` argument to the function and print a deprecation warning if this feature is used. Or am I missing something?

christophwelling, 2022-07-25:
I guess we could still pass the default station and default channel parameters to the detector and give out a deprecation warning. Then it should not break anything.

shallmann, 2022-09-08:
Looking at this just now. In the `GenericDetector` the proposed deprecation warning is already what we do. https://github.com/nu-radio/NuRadioMC/blob/d61a0cd9b32759143347f4b96acc045741db3139/NuRadioReco/detector/generic_detector.py#L102

And I recall that we ran into this issue already in the Gen2 sims, so it would be good to fix. The way it is implemented the generic detector should be backwards compatible. Whereever `reference_station` and `reference_channel` are set these are taken, the fallback is taking the `default_station/channel` passed with the function.
 An option would be to provide a simple function in the generic_detector that checks if a detector is `generic`