nu-radio/NuRadioMC #442: Not correctly merging secondaries in NuRadioProposal?

fschlueter, 2022-09-13:
Hi all,
I found the enclosed code in NuRadioProposal.py and I was wondering if it is correct. `shower_inducing_prods` is the output of `self.__filter_secondaries`. If I understand correctly is `distance` the distance between the secondary and the (parent) lepton (= decay vertex?). Why should if working going from the end of the list? Why should one compare the distances of two secondaries to the vertex to find secondaries which are close to each other? Why going from the back to the front and stop once particles are not close to each other, is the list already somehow sorted?
```
            # group shower-inducing decay products so that they create a single shower
            min_distance = 0.1 * units.m
            while(len(shower_inducing_prods) > 1 and
                   np.abs(shower_inducing_prods[-1].distance - shower_inducing_prods[-2].distance) < min_distance):

                last_decay_prod = shower_inducing_prods.pop(-1)
                shower_inducing_prods[-1].energy += last_decay_prod.energy
                shower_inducing_prods[-1].code = 86
                shower_inducing_prods[-1].name = particle_name[86]

```

daniel-zid, 2022-09-13:
Hi Felix,

It's been a while since I left research, but I'll try to answer your questions as best as possible.

>Why should if working going from the end of the list? 

The code snippet you're looking at has been written for grouping particles coming from a decay into a group of particles that we called a 'decay bundle'. The decay is the last event to happen during tau propagation, so it makes sense that we start from the end. This code takes a look at every shower-inducing particle near the last shower-inducing particle (which could come from a decay), packs them into a bundle, and then treats the bundle as the primary particle for a hadronic cascade.

> Why should one compare the distances of two secondaries to the vertex to find secondaries which are close to each other? 

If I recall correctly, I decided to keep only the distances to save memory. At these high energies, the geometry of the problem is almost a straight line, so if the distance to the vertex is known the position of each secondary is known with high accuracy as well.

>Why going from the back to the front and stop once particles are not close to each other, is the list already somehow sorted?

It should be sorted, or at least that was the case when the code was written. Keep in mind that we are only interested in the decay particles. The code starts with the last shower-inducing particle. If there is another particle within 10 cm of this last particle, it most likely comes from the same decay event, so the last element is popped and a decay bundle is created by adding the energies of the two shower inducing particles. This is not guaranteed, but it would be extremely rare to have two secondaries at the end of the propagation line within 10 cm that don't belong to a decay (if you want to be safe, just ask for much less than 10 cm). More shower-inducing particles may come from the same decay, so the process is repeated. As soon as the next particle in the back of the list is located at or more than 10 cm away from the decay point, the loop is broken and we have effectively transformed all the particles near the decay point into a single decay bundle.

>             # group shower-inducing decay products so that they create a single shower
>             min_distance = 0.1 * units.m

10 cm away from vertex point

>             while(len(shower_inducing_prods) > 1 and
>                    np.abs(shower_inducing_prods[-1].distance - shower_inducing_prods[-2].distance) < min_distance):

While the list has more than one item and the last two items are closer than 10 cm

>                 last_decay_prod = shower_inducing_prods.pop(-1)

Pop the last item
>                 shower_inducing_prods[-1].energy += last_decay_prod.energy

Add the energy from the popped item to the last item in the new list

>                 shower_inducing_prods[-1].code = 86

Enforce code 86 for decay bundles

>                 shower_inducing_prods[-1].name = particle_name[86]

Call it a decay bundle

And that's it! Is there a better way of handling it? Probably, but at the time this worked fine for tau decays. I admit that the code could use some refactoring, and perhaps it could be executed only for tau leptons, which was its main goal.

Important note: due to the way proposal and this adaptation module work, if there is more than one shower-inducing particle from tau decay, these particles are always hadrons, so the bundle is always hadronic. If there is only one particle near the decay point, it is treated as hadronic or electromagnetic according to its type.

fschlueter, 2022-09-13:
Hi Daniel, thanks for your very quick answer, especially considering you left!

> At these high energies, the geometry of the problem is almost a straight line, so if the distance to the vertex is known the position of each secondary is known with high accuracy as well.
Makes sense, I forgot about this. 

Naively I was thinking when the last two particles are not close enough to each other (although other particles are) the while loop is stopped immediately and no bundles are created (an arrangement like [10cm, 15cm, 30cm]).

fschlueter, 2023-01-22:
I spoke once more with @Jean1995 (thanks a lot for this). What I seem to not have grasp is that this function should just combine/merge decay products. Daniel actually writes exactly that, for some reason I did not understand it before...  

And by the way the particles are ordered in time/distance...

@cg-laser, @shallmann, this raises for me the question: Why do we not also merge other showers which are within 10 cm (or a certain distance which is smaller than the wavelength when projecting it in the direction of the Cherenkov angle)? How are showers which are close to each treated? On the Cherenkov ring they should all arrive almost simultaneously?

(I close that issue because its not an issue, would still be interested in discussing that)

closed 2023-01-22 by fschlueter