nu-radio/NuRadioMC #461: NuRadioProposal .pop() in ordered indices looks wrong?

shallmann, 2022-11-02:
https://github.com/nu-radio/NuRadioMC/blob/871a625519eb43f21bbfb527271008edb621ece2/NuRadioMC/EvtGen/NuRadioProposal.py#L622

Maybe not an issue, because there are not multiple decay muons to be propagated per particle? Otherwise popping indices [1,2] as in the loop defined e.g in [0,1,2,3,4] would pop [1,3] rather than [1,2] ...

Jean1995, 2022-11-03:
Good catch. So it should probably look something like

`decay_products = np.delete(decay_products, repropagated_muon_indices).tolist()`

(There is probably a nicer way to write this line, I just have to think a bit more about it 😅 )

shallmann, 2022-11-03:
or reverse ordering indices before popping would also be sufficient ;)

shallmann, 2022-11-04:
accounted for in PR #458

fschlueter, 2023-02-22:
can that be closed?

shallmann, 2023-02-22:
This has been fixed

closed 2023-02-22 by shallmann