nu-radio/NuRadioMC #462: default argument in voltageToEfield converter modules

shallmann, 2022-11-04:
Is there a reason, why the default value from `use_channels=None` is silently overwritten to [0,1,2,3] (I guess this is probably ARIANNA heritage?)

https://github.com/nu-radio/NuRadioMC/blob/3550584f62769a91afe62d645ca71718d47b1e4e/NuRadioReco/modules/voltageToAnalyticEfieldConverter.py#L297
https://github.com/nu-radio/NuRadioMC/blob/3550584f62769a91afe62d645ca71718d47b1e4e/NuRadioReco/modules/voltageToEfieldConverter.py#L148

1. This behaviour looks potentially dangerous
2. (same is maybe true for the set bandpass in the AnalyticEfieldConverter)?
3.  Also the voltageToEfieldConverter fails with a hard-to-track error message, if only called with single antenna in `use_channels`. Since I don't necessarily expect the module does return anything useful in that case, we should either add some logging hint, or require > 1? Any preference?

christophwelling, 2022-11-04:
Yes, I'm pretty sure the default `[0, 1, 2, 3]` is a leftover from ARIANNA.
If you have a default that is an object (like an array), the recommended way to handle it is actually to have `None` in the function definition and then overwrite it with the actual default. If you specify the array right in the function definition, this will create an array object, which is then used as the default. The problem is that on subsequent function calls, this array object will not be recreated, so any changes that are done to the array will change the default the next time the function is called.
In this case, I would suggest to just not have a default for `use_channels`. For the efieldToAnalyticVoltageConverter, it is more complicated, because the `use_channels` parameter has other parameters with defaults before it. So we would either have to remove the defaults for those too or change the order of the parameters. Both will probably break things.
I can check why exactly the module fails if you only have one channel and put in a warning.

fschlueter, 2025-08-08:
I stumbled across that a couple of time too and do not like it neither. My suggestion would be to keep the default to `None` but instead of silently set it to `[0, 1, 2, 3]` raise an error with an appropiate error message that this argument now has to be set. This makes the argument defacto positional but keeps backwards compatibility

sjoerd-bouma, 2025-08-11:
> I stumbled across that a couple of time too and do not like it neither. My suggestion would be to keep the default to `None` but instead of silently set it to `[0, 1, 2, 3]` raise an error with an [...]

I think making this a required argument makes a lot of sense, and raising a warning in the way you suggest maintains backwards compatibility, so this sounds like a good suggestion to me.