nu-radio/NuRadioMC #464: 2D raytracer inheritance

christophwelling, 2022-11-14:
Is there a good reason why the `ray_tracing_2D` class inherits from `ray_tracing_base` ?
I don't think this makes sense, because the 2D raytracer does not follow the general raytracer conventions that is assumed for the `ray_tracing_base` class. For example, the `find_solutions` call works differently (you pass start and end point directly to the method, instead of setting it in another method first) and some internal parameters, like `self._results` are not used.

fschlueter, 2026-09-04:
@hanneswarnhofer, do you agree?

hanneswarnhofer, 2026-09-07:
Yes I agree! The 2D classes don't follow the structure that the `ray_tracing_base` expects, but have some methods that are called the same as ones in the 3D base class, which - if anything - might only be misleading when going through the code. I think the `ray_tracing_2D` and `multi_layer_ray_tracing_2D` classes don't need to inherit from another class.