nu-radio/NuRadioMC #470: Analytic ray-tracing-type disagreement

colemanalan, 2022-11-28:
**Describe the issue**
While running a simulation of a basic RNO-g station, the process crashes during ``get_focusing``.
I honestly find what this function is doing a bit confusing (better commenting might be necessary as well), but  there is an internal ray tracer ``self.__r2d`` which is compared to a secondary ray tracer ``self.__r1``. The solution types seem to not agree which give an error:
`ERROR:ray_tracing_analytic:solution types are not the same`

Relevant function can be found [here](https://github.com/nu-radio/NuRadioMC/blob/develop/NuRadioMC/SignalProp/analyticraytracing.py#L1918).

**Expected behavior**
I guess that it doesn't crash.

**Additional context**
Basic simulation stuff below:

    input_data = generator.generate_eventlist_cylinder(
        "on-the-fly",
        1e3,
        nu_energy,
        nu_energy_max,
        volume,
        thetamin=0.0 * units.rad,
        thetamax=np.pi * units.rad,
        phimin=0.0 * units.rad,
        phimax=2 * np.pi * units.rad,
        start_event_id=1,
        flavor=flavor_ids[flavor],
        n_events_per_file=None,
        spectrum="log_uniform",
        deposited=True,
        proposal=False,
        proposal_config="SouthPole",
        start_file_id=0,
        log_level=None,
        proposal_kwargs={},
        max_n_events_batch=1e3,
        write_events=False,
        seed=root_seed + iSim,
        interaction_type="ccnc",
    )

    sim = mySimulation(
        inputfilename=input_data,
        outputfilename=output_filename,
        detectorfile=detectordescription,
        outputfilenameNuRadioReco=output_filename + ".nur",
        config_file=config,
    )

sjoerd-bouma, 2022-11-29:
This should be *very* rare... is it reproducible? Ideally, I'd need to know the start (=vertex) and end (=antenna position) point of the ray tracing. (Alternatively, if you can write out the offending generated hdf5 file and it contains few enough events, I can check).

In order to compute the focusing correction (see appendix C.6 of the NuRadioMC paper), we need to compute the derivative dTheta/dz. This is done numerically by computing by shifting the receiver position down by 1 cm. In theory, I suppose you could be exactly at the boundary between two ray types. It that really is the case, you just got very unlucky. (Incidentally, it would be possible to fix this edge case if we sort our solutions consistently - this is the aim of #450)

colemanalan, 2022-11-30:
Hi @sjoerd-bouma. Thanks for the comment. However, this is apparently not as rare as you think! I am not really sure I have the intuition here to really know what I am seeing so I will leave it up to you to decide if this is consistent with your hypothesis.

I added some code to print out all the status of the local variables in the function right before it throws the error message, specifically:

> if(self.get_solution_type(iS) != self._r1.get_solution_type(iS)):
    print("BEGIN ERROR DUMP FROM get_focusing")
    print("self.get_receive_vector(iS)", self.get_receive_vector(iS))
    print("recAng", recAng)
    print("lauAng", lauAng)
    print("self.get_launch_vector(iS)", self.get_launch_vector(iS))
    print("self.get_path_length(iS)", self.get_path_length(iS))
    print("self._swap", self._swap)
    print("self._X1", self._X1)
    print("self._X2", self._X2)
    print("vetPos", vetPos)
    print("recPos", recPos)
    print("recPos1", recPos1)
    print("self._r1.get_number_of_solutions()", self._r1.get_number_of_solutions())
    print("lauVec1", lauVec1)
    print("lauAng1", lauAng1)
    print("focusing", focusing)
    print("self.get_solution_type(iS)", self.get_solution_type(iS))
    print("self._r1.get_solution_type(iS)", self._r1.get_solution_type(iS))
    self.__logger.error("solution types are not the same")

Here are a few outputs from this where it failed, but I have probably hundreds more examples where this failed so let me know if you need more.

> BEGIN ERROR DUMP FROM get_focusing
self.get_receive_vector(iS) [ 7.71987810e-01 -6.35637334e-01  2.14006445e-05]
recAng 1.570817727439371
lauAng 0.8811894164363971
self.get_launch_vector(iS) [-0.59558564  0.49039177  0.63623396]
self.get_path_length(iS) 1101.4051993626595
self._swap False
self._X1 [ 690.40771105 -571.46612171 -623.29042021]
self._X2 [ 0. -3. -3.]
vetPos [ 690.40771105 -571.46612171 -623.29042021]
recPos [ 0. -3. -3.]
recPos1 [ 0.   -3.   -3.01]
self._r1.get_number_of_solutions() 2
lauVec1 [-0.59560853  0.49041062  0.636198  ]
lauAng1 0.8812360300675419
focusing 2.2658441189937912
self.get_solution_type(iS) 2
self._r1.get_solution_type(iS) 1

> BEGIN ERROR DUMP FROM get_focusing
self.get_receive_vector(iS) [-7.95963415e-01  6.05344728e-01  3.26238780e-05]
recAng 1.5708289506729143
lauAng 0.8812062874412678
self.get_launch_vector(iS) [ 0.61409127 -0.46702764  0.63622095]
self.get_path_length(iS) 1080.1679578281392
self._swap False
self._X1 [-695.81192991  531.45924772 -609.77684396]
self._X2 [ 3.  0. -3.]
vetPos [-695.81192991  531.45924772 -609.77684396]
recPos [ 3.  0. -3.]
recPos1 [ 3.    0.   -3.01]
self._r1.get_number_of_solutions() 2
lauVec1 [ 0.61411488 -0.46704559  0.63618498]
lauAng1 0.8812529129272949
focusing 2.244178158066147
self.get_solution_type(iS) 2
self._r1.get_solution_type(iS) 1

> BEGIN ERROR DUMP FROM get_focusing
self.get_receive_vector(iS) [ 9.37402182e-01 -3.48248680e-01  8.38747255e-05]
recAng 1.5708802015205023
lauAng 0.8816946256818422
self.get_launch_vector(iS) [-0.72350341  0.26878443  0.63584412]
self.get_path_length(iS) 870.4364850486329
self._swap False
self._X1 [ 671.2874259  -252.38597785 -476.36145279]
self._X2 [ 0. -3. -3.]
vetPos [ 671.2874259  -252.38597785 -476.36145279]
recPos [ 0. -3. -3.]
recPos1 [ 0.   -3.   -3.01]
self._r1.get_number_of_solutions() 2
lauVec1 [-0.72353125  0.26879477  0.63580807]
lauAng1 0.881741324778113
focusing 2.016149730447698
self.get_solution_type(iS) 2
self._r1.get_solution_type(iS) 1

> BEGIN ERROR DUMP FROM get_focusing
self.get_receive_vector(iS) [6.74067676e-01 7.38669594e-01 1.77753616e-05]
recAng 1.5708141021565316
lauAng 0.8813751315973125
self.get_launch_vector(iS) [-0.52012027 -0.56996803  0.63609067]
self.get_path_length(iS) 964.0259754022671
self._swap False
self._X1 [ 528.3898848   582.31771747 -535.89370877]
self._X2 [-3.  0. -3.]
vetPos [ 528.3898848   582.31771747 -535.89370877]
recPos [-3.  0. -3.]
recPos1 [-3.    0.   -3.01]
self._r1.get_number_of_solutions() 2
lauVec1 [-0.52014027 -0.56998995  0.63605468]
lauAng1 0.8814217763397223
focusing 2.1205363308862517
self.get_solution_type(iS) 2
self._r1.get_solution_type(iS) 1

sjoerd-bouma, 2022-11-30:
Thanks, this is very helpful. I'm not sure which ice model you used (I'm guessing it's **not** `greenland_simple`, which doesn't return any solutions at all for the given geometries), but I was able to reproduce the error with `southpole_2015`.

It *seems* that all of the examples you've provided are rays which are very close to horizontal at the receiver. We currently classify ray types depending on the angle they're received at ('direct' = downgoing, 'refracted' = upgoing), leading to this error message. In my opinion, this classification is not very meaningful anyway, and after #450 I think we can safely dispense with this error message. It's still an interesting curiosity that you get so many of them - I would have expected them to be very rare.

For now, I think you can safely ignore the error message (despite it being labelled 'error', it doesn't actually raise an error, so it shouldn't crash your simulations).

sjoerd-bouma, 2023-02-16:
I'm closing this as the relevant error message was removed in #450, but feel free to run some tests and reopen this should something unexpected still be happening!

closed 2023-02-16 by sjoerd-bouma