nu-radio/NuRadioMC #490: Veff utility only effectively counts the last file read in

colemanalan, 2023-02-13:
In the `get_Veff_Aeff_array`, there is an attempt to split the `data` block into energy, zenith angle, and trigger bins.
However, all this effectively does is assign the value of the last entry in `data`. There is no sum over the various entries.

To replicate this bug:
 Have two hdf files with the same energy range and zenith range. Read in both the files using the `get_Veff` function and then read in only the last/second of the two files and print out the `output` that is returned by this function. They will be the same.

To fix this, this line should probably be changed to some kind of sum, depending on the field (i.e. you can add the weight sum, but you should linearly add the errors).
https://github.com/nu-radio/NuRadioMC/blob/6ab5c8cbdb53b02d7d2fb15847bf47fc21bf8588/NuRadioMC/utilities/Veff.py#L713

cg-laser, 2023-02-13:
can you post how exactly you are opening the files and combining them?

colemanalan, 2023-02-13:
> input_dir = "/some/dir/with/hdf/files/"
> data = get_Veff_Aeff(input_dir)
> Veffs, energies, energies_low, energies_up, zenith_bins, utrigger_names = get_Veff_Aeff_array(data)
> print(Veffs)

> one_file = "/some/dir/with/hdf/files/file_999.hdf"
> data = get_Veff_Aeff(one_file)
> Veffs, energies, energies_low, energies_up, zenith_bins, utrigger_names = get_Veff_Aeff_array(data)
> print(Veffs)

`Veffs` will be the same in both cases

This is how it is recommended to use these functions in the docstrings for these two functions as well as `T03visualizeVeff.py`

cg-laser, 2023-02-13:
thanks, yes this should be fixed or caught. The intended behavior is to first merge hdf5 files with the same simulation settings together using the `merge_hdf5.py` tool.

fschlueter, 2025-05-02:
Is this fixed?