nu-radio/NuRadioMC #494: `proposal` cannot be pip installed

sjoerd-bouma, 2023-02-22:
This is probably due to one of its dependencies (`conan`) having recently been updated. I think we'll have to wait for the proposal devs to update proposal to be installable with the new conan, or to fix their conan version to 1.59.0. See [this issue](https://github.com/tudo-astroparticlephysics/PROPOSAL/issues/354).

In the meantime, I've created a fork with conan fixed to <2.0, so proposal 7.5.0 can be installed using `pip install git+https://github.com/sjoerd-bouma/PROPOSAL`

fschlueter, 2023-02-23:
What should we do @cg-laser, @shallmann? Should we wait until proposal fixes their problem or should we temporarily switch to sjoerd's fork? And in the meant time should we continue merging stuff a few of the PR have successful pipelines.

shallmann, 2023-02-23:
Would it be possible to pip install a fixed Conan version prior to proposal?

shallmann, 2023-02-23:
Or would that be upgraded when pip installing proposal afterwards?

sjoerd-bouma, 2023-02-23:
@fschlueter I think we can wait, it's already been fixed and approved by the proposal devs - as soon as it gets pushed to pip we can just upgrade proposal.
I reckon in the meantime anything that doesn't mess with proposal and passes all the other tests can safely be merged.

@shallmann unfortunately, that doesn't work. But my fork or now the `pin_conan` branch on proposal can also be pip installed, so I think that's straightforward enough as  a workaround for the next couple of days.

sjoerd-bouma, 2023-02-23:
i.e. `pip install git+https://github.com/tudo-astroparticlephysics/PROPOSAL` should work as well, now.

shallmann, 2023-02-23:
Thanks Sjoerd for explaining. So it seems we can just wait 1 or 2 days and use the 7.5.1 or whatever version

PR #510 (2023-03-10): Update PROPOSAL to version 7.5.1
This updates PROPOSAL to version 7.5.1.
This fixes #494, so we don't have to checkout a specific commit anymore.

All other bugfixes included in this patch release of PROPOSAL don't include any changes that are relevant for NuRadioMC (see [here](https://github.com/tudo-astroparticlephysics/PROPOSAL/releases/tag/7.5.1) for the complete Changelog).

Should I also update the `changelog.txt` for this?
closed 2023-03-10 by shallmann