nu-radio/NuRadioMC #495: Measured noise importer

fschlueter, 2023-02-23:
Hi all,

I opened this issue to have a discussion about the various noise importer for measured noise we have. From what I see have we have:

- `io/noise/noiseImporter.py`
- `io/noise/noiseImporterROOT.py`
- `channelMeasuredNoiseAdder.py`

Furthermore, #293 implements a `io/noise/noiseImporterUproot.py` which (afaik) is suppose to add noise for/from the ARIANNA detector. I am currently working on a module to import noise from/for RNO-G. My question is now: Do we need the above modules, are some maybe redundant (in particular the `io/noise/noiseImporter.py` and `channelMeasuredNoiseAdder.py`) ? What I imagine could be necessary is the following: We have one generic noise adder module which uses `.nur` files to import noise and than modules for the specific detectors which use their specific data formats and are call appropriately with the detector name. Does this make sense?

PS: If we keep the module, we should move `channelMeasuredNoiseAdder.py` to `io/noise/`

cg-laser, 2023-02-23:
short answer is yes ;-)

all of them do slightly different things. The three modules in `io/noise` read noise files from different sources. 
The `channelMeasuredNoiseAdder.py` assumes (but @christophwelling should comment in case I got this wrong) that the station ids and channels ids between the simulated stations and datafiles are the same. Hence, this can directly be used e.g. for RNO-G data if the RNO-G data is available in nur files. 

But I agree, it is confusing that some are under `io/noise` and others are directly under `modules`. Normally, things in `io` should "yield" a new event and not modify an existing one. So all these modules add noise to existing channels. 

I don't think it is practical to generalize everything in one module, but we should rename the modules and improve the doc string to make clear what each module does.

fschlueter, 2023-02-24:
I agree, I would not generalise it neither. So should we create a new directory in `modules/` which is called `noise` or `measured_noise` and place the modules there? Furthermore should we rename the modules `noiseImporter.py` -> `noiseImporterARIANNA.py` and `noiseImporterROOT.py` -> `noiseImporterROOTARIANNA.py` (if that are appropiate names?)

cg-laser, 2023-03-07:
the suggestion sounds good to me! But I would use `noiseImporterARIANNAROOT` instead of `noiseImporterROOTARIANNA`

fschlueter, 2023-04-01:
I close this issue, it will be addressed very soon with the merging of #508

closed 2023-04-01 by fschlueter