nu-radio/NuRadioMC #500: Simulation of ADC expects symmetric voltage to ADC relation

fschlueter, 2023-02-27:
The title of the issue might be confusing, sorry for that.

I have the suspicion that the following line is dangerous: 

**NuRadioReco/modules/analogToDigitalConverter.py:40**
```
lsb_voltage = adc_ref_voltage / (2 ** (adc_n_bits - 1) - 1)
```
with `ads_ref_voltage` being:
```
    adc_ref_voltage: float
        Voltage corresponding to the maximum number of counts given by the
        ADC: 2**adc_n_bits - 1
```
However, this is only true if voltage_max = - voltage_min, correct? I.e. if that holds:
```
v_ref_peak_to_peak = (voltage_max - voltage_min) = 2 voltage_max 
lsb_voltage = v_ref_peak_to_peak / (2 ** (adc_n_bits) - 1) = voltage_max / (2 ** (adc_n_bits - 1) - 1)
```
(compare the exponents).

cg-laser, 2023-03-07:
I think in general you are correct but this is so far a "simulation only" module. If we want to model, e.g., RNO-G in the future, we would plug in the measured voltage to ADC conversion. So in the future, one could add the option to read the conversion from some calibration table/database etc. 
I don't think it would be useful to just add an `offset` parameter to this module.

fschlueter, 2023-03-07:
Agree, but why not changing the code to use:
```
v_ref_peak_to_peak / (2 ** (adc_n_bits) - 1)
```
This is also correct with an offset (?). After all an offset, i.e., a pedestal is typically removed.

fschlueter, 2024-11-18:
I stumbled across this last week again. I am feeling that this will cause a bug sooner or later. For example If I understand the code correct,  to simulate the RADIANT response correctly we want to simulate the effect of an 12-bit digitiser with a range of 2.5V. However, with the current code we would need to set the reference voltage to 1.25V correct? This is awkward.

ryankrebs016, 2025-01-21:
Bringing this convo from the above PR to here. 

First the adc_ref_voltage is somewhat awkward and confusing to use. The analogToDigitalConverter.py module calculates the ref voltage using `adc_ref_voltage = Vrms * (2^(adc_n_bits-1)-1) / (2^(adc_noise_n_bits-1)-1)` and then uses the `lsb_voltage=adc_ref_voltage/(2^(adc_n_bits-1)-1)` to do the ceiling/flooring. `(2^(adc_n_bits-1)-1)` is canceled out in the calculation and it feels wrong to use as it describes the number of levels for only one half of the adc's range with an extra -1 (which I think is a bug). I'm only finding sources online that have the reference voltage as Vmax-Vmin. Using the FLOWER board as an example, the input range is 2V, from -1V to 1V, on an 8 bit ADC, so it's correct to have the ref_voltage be the 2V and then have the total number of steps as `2^(adc_n_bits)`. The flooring/ceiling then takes care of properly representing the max and min voltage values it digitizes.

ryankrebs016, 2025-01-21:
Second, the noise level being defined as number of bits is somewhat ambiguous, one could conceivably use `noise_counts=2^(adc_noise_n_bits-1) -1`,  `noise_counts=2^(adc_noise_n_bits-1)`, or `noise_counts=2^(adc_noise_n_bits)` depending on how the calculation is done in the code. For example, https://github.com/nu-radio/NuRadioMC/blob/4e2b47fc5485255e03f069f4fe778c12392c8287/NuRadioReco/modules/analogToDigitalConverter.py#L293 vs https://github.com/nu-radio/NuRadioMC/blob/7073bdbc08280995513e8db499f29c7c5fbbf6d9/NuRadioReco/modules/RNO_G/triggerBoardResponse.py#L155 . The best method would be to define the adc counts or voltage level explicitly.

fschlueter, 2025-01-23:
I am in principle also in favor of changing this. However, this would mean that we I) have to change all existing detector descriptions which are using this convention and II) risk that we introduce a bug in the user code of people not noticing this change and not adapting their detector description. 

However, I wonder how often this might cause a bug because people imply a different convention.

fschlueter, 2025-05-02:
This is implemented by now

closed 2025-05-02 by fschlueter