nu-radio/NuRadioMC #504: Extending the ARZ shower library

fschlueter, 2023-03-01:
@christophwelling suggested to extend the library of shower used by the ARZ model to produce neutrino radio signals. Currently we have 10 shower profiles per shower type (EM, HAD) and energy bin. Kaeli produced showers in the PUEO context which we could use. I have converted the showers in the appropriate format and stored them in a pickle file which should be readable by NuRadioMC (Not explicitly tested yet). I have produced some plots to compare those shower profiles to the one currently used by NuRadioMC (labelled in the following plots with "v1.2"). 

The plots raise quite some questions:

- It seems like that the new EM showers are not simulated taking into account the LPM effect
- The new showers seem to be shifted (by a constant shift?!) in depth
- For the hadronic showers the amplitude of the charge excess seems incompatible, in particular at lower energies

A comment: To be able to compare the two different shower libraries I rescaled the amplitudes linear with energy (because the sets have not been generated with the same energy bins) 

  
[energybins.pdf](https://github.com/nu-radio/NuRadioMC/files/10862667/energybins.pdf)
[profiles_EM_1.00e+16.pdf](https://github.com/nu-radio/NuRadioMC/files/10862668/profiles_EM_1.00e%2B16.pdf)
[profiles_EM_1.00e+17.pdf](https://github.com/nu-radio/NuRadioMC/files/10862669/profiles_EM_1.00e%2B17.pdf)
[profiles_EM_1.00e+18.pdf](https://github.com/nu-radio/NuRadioMC/files/10862670/profiles_EM_1.00e%2B18.pdf)
[profiles_EM_1.00e+19.pdf](https://github.com/nu-radio/NuRadioMC/files/10862671/profiles_EM_1.00e%2B19.pdf)
[profiles_EM_1.00e+20.pdf](https://github.com/nu-radio/NuRadioMC/files/10862672/profiles_EM_1.00e%2B20.pdf)
[profiles_HAD_1.00e+16.pdf](https://github.com/nu-radio/NuRadioMC/files/10862673/profiles_HAD_1.00e%2B16.pdf)
[profiles_HAD_1.00e+17.pdf](https://github.com/nu-radio/NuRadioMC/files/10862674/profiles_HAD_1.00e%2B17.pdf)
[profiles_HAD_1.00e+18.pdf](https://github.com/nu-radio/NuRadioMC/files/10862675/profiles_HAD_1.00e%2B18.pdf)
[profiles_HAD_1.00e+19.pdf](https://github.com/nu-radio/NuRadioMC/files/10862676/profiles_HAD_1.00e%2B19.pdf)
[profiles_HAD_1.00e+20.pdf](https://github.com/nu-radio/NuRadioMC/files/10862677/profiles_HAD_1.00e%2B20.pdf)

christophwelling, 2023-03-01:
The old EM shower at 1e20eV also looks like it's clipped at the end.

fschlueter, 2023-03-02:
> The old EM shower at 1e20eV also looks like it's clipped at the end.

Indeed.

shallmann, 2023-03-02:
(re to Christophs comment)
-  I quickly looked, and in the 1.2 shower library it seems the depth goes to 20k g/cm**2, so the clipping is perhaps not an issue in the shower library itself but just plotting?
- I also checked 2 weeks ago because I was curious how often we would get very far away secondary LPM showers for nueCC with proposal (proposal can also propagate electrons) ... for most energies I think it is not relevant since only very rarely secondary showers are far from the primary interaction, but we could e.g. use proposal to propagate 1e20 if need be (I.e. I could make a PR out of my modified generator code if people think it is useful)? The attached plots are simulating a number (10k) 1e19 and 1e20 primaries in infinite ice and look at 
(a) the cumulative distribution of showers > certain energies as a function of distance from the primary interaction. - y-axis is number of showers per interaction 
(b) 2 d distribution of shower energy and distance, again normalised to per injected primary.

[1d_emins_1.00e+19.pdf](https://github.com/nu-radio/NuRadioMC/files/10869302/1d_emins_1.00e%2B19.pdf)
[1d_emins_1.00e+20.pdf](https://github.com/nu-radio/NuRadioMC/files/10869303/1d_emins_1.00e%2B20.pdf)
[2d_1.00e+19.pdf](https://github.com/nu-radio/NuRadioMC/files/10869305/2d_1.00e%2B19.pdf)
[2d_1.00e+20.pdf](https://github.com/nu-radio/NuRadioMC/files/10869304/2d_1.00e%2B20.pdf)

fschlueter, 2023-03-02:
@shallmann is right, I set the range of the plot (sorry did not catch this when I checked the code earlier this morning).

shallmann, 2023-03-02:
And concerning your above questions, I think I agree with your three points, maybe @cg-laser has more comments. @KaeliAutumn it would be interesting to understand why they are shifted/disagree (the 1.2 ones clearly look more like what we expect, right?)... In general we do want to have LPM enabled I'd say.

Is it a lot of effort to extend the shower library with the old scripts that we had (did not check: is the code there?)?

fschlueter, 2023-03-02:
@shallmann, what means "shower"? Each primary produces one shower and the position of the shower is defined by the maximum of the charge-excess profile / energy deposit, or can one primary produce several showers and one identifies several peaks in a profile which are sufficiently displaced from each other?

shallmann, 2023-03-02:
> @shallmann is right, I set the range of the plot (sorry did not catch this when I checked the code earlier this morning).

For 1e20 and beyond it might (and probably should) however still be not enough to have the 200 g/cm**2 according to this plot I found (caveat: I don't remember where this plot came from originally, I think it might have been SalSA, i.e. perhaps a density more like 2.2 g/cm**3 -> i.e. for ice it is possible that one needs to rescale y by ~2) ... but the general behaviour with energy should be clear from this
[image]

shallmann, 2023-03-02:
> @shallmann, what means "shower"? Each primary produces one shower and the position of the shower is defined by the maximum of the charge-excess profile / energy deposit, or can one primary produce [...]

I injected electrons (!) (PDG-id 11) with 1e19 and 1e20 eV, and collected all catastrophic secondary emissions that proposal returned. I ran 10k events and averaged over them. ... i.e. you can read the 1D plots as follows: for a 1e20 electron, on average one expects 1 shower with >1e17 eV (orange line) having a distance of 200m or more from the starting point