nu-radio/NuRadioMC #524: Defining trigger time in NuRadioReco

fschlueter, 2023-03-31:
Should it be defined absolut or relative (to station time). In what format do we want to store it (astropy, float tuple, ...). Do we need ns precision?

sjoerd-bouma, 2023-04-06:
Some more comments:
- we currently store astropy.time.Time objects by calling `.value` (see https://github.com/nu-radio/NuRadioMC/blob/9b588ee8ba3cea753b30086652c6ec22f78dcffb/NuRadioReco/framework/base_station.py#L241-L247). I'm not sure if this ensures the precision we're interested in. E.g. trying to set an astropy.time.Time object with a `np.float128`:
```
from astropy.time import Time
import numpy as np
time_128 = np.float128(int(Time.now().unix)) + np.float128(1e-9)
print(time_128, type(time_128))
aptime = Time(time_128, format='unix')
print(aptime.value, type(aptime.value))
```
results in the output
```
1680768370.000000001 <class 'numpy.float128'>
1680768370.0 <class 'numpy.float64'>
```
If we do store our times as unix doubles, the precision is limited to [52 bits](https://numpy.org/doc/stable/user/basics.types.html#array-types-and-conversions-between-types), which means ~ 400 ns. This will limit what we can do with multiple-station coincidences.

fschlueter, 2023-04-28:
Just to put the information here and to restart the discussion. For the (just merged) mattak reader we set the trigger time absolute and the traces start time is relative to that.

fschlueter, 2023-05-04:
> Some more comments:
>
> * we currently store astropy.time.Time objects by calling `.value` (see [...]

Good that you bring that up Sjoerd. So we already store the time(stamp) as a float to file. I think that is the right way to go because storing astropy objects carries the danger of compatibility problems (e.g. see https://github.com/nu-radio/NuRadioMC/issues/481 where this happened for the event header where we used to store astropy objects). I would suggest the following: Within NuRadioMC we use astropy where ns precision is required and to file we store two floats/integers (the second and the nanosecond).

fschlueter, 2025-02-11:
We can close this Issue. It should have been addressed with #798 and #773

closed 2025-02-11 by fschlueter