nu-radio/NuRadioMC #527: Cleanup voltageToEfieldConverter

cg-laser, 2023-04-20:
The voltageToEfieldConverter needs some cleanup. 
The solution to the electric field for two antennas is calculated twice and never used: https://github.com/nu-radio/NuRadioMC/blob/develop/NuRadioReco/modules/voltageToEfieldConverter.py#L161-L171

Intended behavior: In case of just 2 antennas, don't use the chi2 minimization but instead us the exact solution to calculate the efield.

MijnheerD, 2023-11-22:
@karenterveer had a look at this module last week, and I think this has now been fixed?

sjoerd-bouma, 2023-11-22:
It was me, actually. The code could be tidied up by removing the lines @cg-laser mentions, but in the case of just 2 antennas, the chi2 solution is equivalent to the exact solution (which I verified), with the only benefit to using a separate exact calculation for that case (that I can see) a small improvement in performance.

This has not been implemented yet but I don't think it's a particularly high priority either.

cg-laser, 2023-11-23:
great that you verified that! I agree, then we don't need to implement a special case. We could add a comment though

sjoerd-bouma, 2024-12-10:
Closed by #758

closed 2024-12-10 by sjoerd-bouma