nu-radio/NuRadioMC #533: Newest versions of numpy (>=1.20.0) do not work

clark2668, 2023-05-07:
**Describe the issue**

Numpy deprecated a bunch of aliases as of 1.20.0. See [here](https://levelup.gitconnected.com/fix-attributeerror-module-numpy-has-no-attribute-float-d7d68c5a4971) for more info. This means that lots of `np.float`s distributed throughout the code (e.g. in the [generator](https://github.com/nu-radio/NuRadioMC/blob/develop/NuRadioMC/EvtGen/generator.py#L1233) throw errors like the one below:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data/i3home/baclark/RNOG/rnog_simulation/tools/miniconda/lib/python3.10/site-packages/numpy/__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations. Did you mean: 'cfloat'?
```

**Expected behavior**
The import should succeed.

**Additional context**
None to add.

clark2668, 2023-05-07:
As an aside, I'm not sure why this wasn't caught by the build-bot. Is the build bot somehow accidentally invoking old python or old numpy?

sjoerd-bouma, 2023-05-07:
Thanks Brian for flagging this. The CI runs on Python 3.7, where pip resolves the dependencies to install numpy 1.21.6, it seems, and np.float has only started raising an error since numpy 1.24. Should of course be fixed anyway.

clark2668, 2023-05-07:
No problem. Thanks for the really rapid response Sjoerd!

sjoerd-bouma, 2023-06-23:
#538 fixed this in develop, but #534 (into `master`) didn't include `np.complex` and `np.bool` yet. We may want to merge #538 into master as well (will have a look next week).

sjoerd-bouma, 2024-01-26:
This is fixed as of v2.2.0

PR #538 (2023-05-19): Hotfix/numpy float
Closes #533 / merges #534 into develop
closed 2024-01-26 by sjoerd-bouma