nu-radio/NuRadioMC #540: RNO-G mattack reader seems to contain bug

thoglu, 2023-05-19:
Just running the example script https://github.com/nu-radio/NuRadioMC/blob/develop/NuRadioReco/examples/RNO_data/read_data_example/read_rnog.py
fails because in 
https://github.com/nu-radio/NuRadioMC/blob/develop/NuRadioReco/modules/io/RNO_G/readRNOGDataMattak.py#LL149C32-L149C40

it wants to access `rnog_data` which does not exist.

fschlueter, 2023-05-19:
Hi @thoglu, this `rnog_data` module lives in a different repository (https://github.com/RNO-G/rnog-runtable). Clone that and put it in your python path and it should work. The code should give you an error msg which tells you to clone a repository. However, currently a wrong repo is named. This is getting fixed in https://github.com/nu-radio/NuRadioMC/pull/539

fschlueter, 2023-05-19:
@shallmann @sjoerd-bouma can we add the rnog-runtable repo as optional dependency to the install script?

thoglu, 2023-05-19:
Hey, there is a second bug in the init (I updated the above description):

If rnog_data fails, it tries to access the logger in 

https://github.com/nu-radio/NuRadioMC/blob/develop/NuRadioReco/modules/io/RNO_G/readRNOGDataMattak.py#L154
or 
https://github.com/nu-radio/NuRadioMC/blob/develop/NuRadioReco/modules/io/RNO_G/readRNOGDataMattak.py#L157.

The logger is not yet defined in init though... it is defined in the 'begin' function.

shallmann, 2023-05-19:
@fschlueter I thought the reader would still work without having `rnog_data` installed? I think it should and the plan was to have the access to the runtable completely optional
 
Maybe @sjoerd-bouma ... I would add a note, that this is an RNO-G private repo

fschlueter, 2023-05-19:
eieiei, sorry for that

fschlueter, 2023-05-19:
that happens when one does not test changes and has no unit testing...

fschlueter, 2023-05-19:
> @fschlueter I thought the reader would still work without having `rnog_data` installed? I think it should and the plan was to have the access to the runtable completely optional
> 
> Maybe @sjoerd-bouma ... I would add a note, that this is an RNO-G private repo

It should. Maybe it is related to the other bug @thoglu found... However, isn't it still a good idea?

sjoerd-bouma, 2023-05-19:
> @shallmann @sjoerd-bouma can we add the rnog-runtable repo as optional dependency to the install script?

This is already done? `rnog_data` can be selected when running the install script; if the user does not have the required permission to clone the repository, I assume the installation will just fail. But I would consider that an RNO-G issue rather than a NuRadioMC one.

fschlueter, 2023-05-19:
ahh right!  you are great @sjoerd-bouma :)

closed 2023-05-19 by fschlueter