nu-radio/NuRadioMC #566: Eventbrowser

fschlueter, 2023-09-01:
calls to `self.logger.debug` in `NuRadioRecoIO` are repeated several times with in the event browser app:

```
DEBUG - NuRadioReco.NuRadioRecoio - Finished scanning file 0.
DEBUG:NuRadioReco.NuRadioRecoio:Finished scanning file 0.
DEBUG - NuRadioReco.NuRadioRecoio - Finished scanning file 0.
DEBUG:NuRadioReco.NuRadioRecoio:Finished scanning file 0.
DEBUG - NuRadioReco.NuRadioRecoio - Finished scanning file 0.
DEBUG:NuRadioReco.NuRadioRecoio:Finished scanning file 0.
DEBUG - NuRadioReco.NuRadioRecoio - Finished scanning file 0.
DEBUG:NuRadioReco.NuRadioRecoio:Finished scanning file 0.
DEBUG - NuRadioReco.NuRadioRecoio - Finished scanning file 0.
DEBUG:NuRadioReco.NuRadioRecoio:Finished scanning file 0.
INFO - NuRadioReco.NuRadioRecoio - ... finished in 1 seconds
INFO:NuRadioReco.NuRadioRecoio:... finished in 1 seconds
INFO - NuRadioReco.NuRadioRecoio - ... finished in 1 seconds
INFO:NuRadioReco.NuRadioRecoio:... finished in 1 seconds
INFO - NuRadioReco.NuRadioRecoio - ... finished in 1 seconds
INFO:NuRadioReco.NuRadioRecoio:... finished in 1 seconds
INFO - NuRadioReco.NuRadioRecoio - ... finished in 1 seconds
INFO:NuRadioReco.NuRadioRecoio:... finished in 1 seconds
INFO - NuRadioReco.NuRadioRecoio - ... finished in 1 seconds
INFO:NuRadioReco.NuRadioRecoio:... finished in 1 seconds
```

Following a thread on stack overflow I tried the following in `app.py` which did not work

```
if app.logger.hasHandlers():
    app.logger.handlers.clear()
```

MijnheerD, 2023-09-07:
I have investigated this issue further, as the event browser keeps crashing on my machine when trying to open a LOFAR event file. What is actually happening is the creation of multiple instances of the `NuRadioRecoio` class. I added a statement to print the output of `DataProvider.get_file_handler()` and could identify at least 4 different instances of `NuRadioRecoio` for the same user ID. 

My guess is that this function is called multiple times in same kind of parallelized manner, resulting in the file handler not being saved in the `DataProvider` instance before the next call is made, resulting in the .nur file being opened multiple times. This is probably causing the massive lag and eventual crash on my system, as the event browser is using up more than 5GB of RAM according to the logs.

sjoerd-bouma, 2023-12-05:
Fixed by #557 (hopefully?)