nu-radio/NuRadioMC #569: proposal doesn't work again (undefined symbol)

sjoerd-bouma, 2023-09-12:
The tests currently fail because of an issue with proposal 7.5.1. The installation of proposal does not raise an error, but subsequently importing anything relying on proposal fails with and `undefined symbol` error. Example below:

```
WARNING: autodoc: failed to import module 'NuRadioProposal' from module 'NuRadioMC.EvtGen'; the following exception was raised:
/opt/hostedtoolcache/Python/3.7.17/x64/lib/python3.7/site-packages/proposal.cpython-37m-x86_64-linux-gnu.so: undefined symbol: _ZN5boost10filesystem4pathdVERKS1_
```
It's not clear to me if this is a python 3.7 exclusive issue or due to some updated dependency of proposal, I can't reproduce this on Ubuntu 22.04 and python 3.10.12. I also do not know of a workaround.

sjoerd-bouma, 2023-09-12:
Perhaps unsurprisingly, considering the issue occurs in a compiled .so object, but no NuRadioMC dependencies nor the Python version have changed between the last successful run and first occurrence of this error. On the other hand, the proposal wheel took much longer to build and has a different size, so my guess would be the issue is caused by some updated proposal dependency.

sjoerd-bouma, 2023-09-12:
Opened an issue on the proposal github: https://github.com/tudo-astroparticlephysics/PROPOSAL/issues/377

Jean1995, 2023-09-13:
Which one is the action where proposal has been re-built and stopped working?

Jean1995, 2023-09-13:
I believe something has changed about the boost package provided by conan. I've just realized that our CI isn't working anymore as well.

PR #570 (2023-10-04): upgrade proposal to version 7.6.2
Upgrade PROPOSAL to version 7.6.2

Notably, this upgrades the package manager `conan` (which is used to install proposal dependencies) from version 1 to version 2. This should increase stability, and hopefully fix #569 

Full release notes of PROPOSAL can be found [here](https://github.com/tudo-astroparticlephysics/PROPOSAL/releases), although none of the other changes should affect NuRadioMC.
closed 2023-10-04 by sjoerd-bouma