nu-radio/NuRadioMC #576: Cannot pip install mysql-python

clark2668, 2023-09-22:
**Describe the issue**
When trying to pip install `mysql-python`, the installation breaks. Somehow `mysql-python` wasn't bild to be compatible with py3?

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

PrathamDubey06, 2023-10-30:
The correct syntax for using MySQL in Python code is to install the mysql-connector library. To do this, you need to install the MySQL Connector library on your system. You can do this by executing the following code in the command prompt (CMD):

`pip install mysql-connector-python`

clark2668, 2023-12-06:
Hi -- thanks! Can the NuRadioMC docs be updated to specify this? As they are right now, this install command doesn't work.

Also, I noticed that `future` and `radiotools` are required packages, but aren't listed in the "one shot" list (https://nu-radio.github.io/NuRadioMC/Introduction/pages/installation.html#pip-installable-dependencies). Can these also be added? Thanks!

[image]

sjoerd-bouma, 2025-02-04:
I think this has been updated in the documentation. It's not currently included in the pyproject.toml but until someone who is actively using this lets me know I might leave it as is.

closed 2025-02-04 by sjoerd-bouma

clark2668, 2025-02-04:
Hi -- great, I'm glad this is fixed! Thank you!