nu-radio/NuRadioMC #582: Time adjuster creates invalid waveforms

colemanalan, 2023-09-27:
When the wavefom is assigned in `triggerTimeAdjuster`:
https://github.com/nu-radio/NuRadioMC/blob/7b5b259c2b06684b32f49c5610a6021e767cdadb/NuRadioReco/modules/triggerTimeAdjuster.py#L96

there is an explicit check when `set_trace` is performed to ensure that waveforms cannot have an odd length. However, this module can generate waveforms with an odd length when cutting when the pulse is too close to the "right-most" edge or more generally when a pre-trigger time is requested that is larger than the waveform. This causes the whole program to crash.

Example:
If this function is called, for a desired waveform length of 1024 bins at 2.4 GS/s but is currently sampled at 5 GS/s, then this module will try to make waveforms that are: 1024 * 5 / 2.4 = 2133 bins long
https://github.com/nu-radio/NuRadioMC/blob/7b5b259c2b06684b32f49c5610a6021e767cdadb/NuRadioReco/modules/triggerTimeAdjuster.py#L63

sjoerd-bouma, 2023-09-28:
I've already fixed this on #526, but I should finally implement the trigger time storage as requested by Christian so that can be merged, apologies.

sjoerd-bouma, 2024-01-26:
Closed by #526

PR #595 (2023-12-22): Release v2.2.0
There have been a large number of bug fixes and improvements added to NuRadioMC since the last release, so it's time for the next one. We discussed yesterday which pull requests we would still like to merge before this release; this PR can serve as a place to track and discuss this.

**Include in 2.2.0**
- [x] #557 (the eventbrowser does not work on the current develop)
~~- [ ] #556 (add initial LOFAR reader to enable LOFAR students to get started)~~ the meaning of group_id will be changed. We should not make a release with that as this will be confusing and a non-backward compa [... cut]
closed 2024-01-26 by sjoerd-bouma