nu-radio/NuRadioMC #592: Reading in a large detector from json is slow

sjoerd-bouma, 2023-11-16:
Reading in a large detector description from a `.json` file is slower than expected; for the LOFAR detector description (O(1000) channels), reading in the json description using
```
det = Detector('LOFAR.json', antenna_by_depth=False)
```
takes O(1 minute). If instead we read in the json first, and then load the detector from the dictionary, this only takes O(1 s):
```
with open('LOFAR.json', 'r') as f:
    det_dict = json.load(f)
det = Detector(json_filename=None, source='dictionary', dictionary=det_dict, antenna_by_depth=False)
```

I'm not sure how the first case is currently implemented - most time is spent in `tinydb_serialization._decode_deep` and `json.decoder.raw_decode`. However, seeing as the second case is (?) equivalent to the first, we should probably change the implementation to just internally read in the json as a dictionary first, and load the detector from that.

MijnheerD, 2023-11-21:
Something I do not entirely get though, is that on lines 95 and 96 of `generic_detector.py` it seems like the JSON file decoded and the `Detector` is constructed using the dictionary on line 97-99. So would the time be spent on the `Query()` calls then?

anelles, 2023-11-21:
If I am not fully mistaken @cg-laser was the one to introduce tiny_db in here. Maybe it really is too tiny for what we are doing here? 
I would be open to changing it, if we can make it work seamlessly with what @jakobhenrichs and @fschlueter have been implementing for the DB-based detector.

fschlueter, 2023-11-21:
@anelles, I do not think so. The RNO-G detector class uses a custom buffering "strategy", I do not think that this is transferable to this detector.

cg-laser, 2023-11-23:
TinyDB is still used when giving a json object to the detector class. I second your suggestion Sjoerd. It seems to efficiently solve the problem.

MijnheerD, 2024-02-21:
As a development on this issue, when loading a complete detector description the initialisation is much faster. So I think this is particular issue when trying to load a GenericDetector with lots of references, which is where I would guess it spends most of its time.

fschlueter, 2025-05-02:
Can this be closed? I think this is a feature of the generic detector class...

MijnheerD, 2025-05-02:
Probably it can. Just this morning I did come across a package that could help us though: https://pypi.org/project/pysimdjson/ It is a library which supposedly speeds up the parsing of JSON files a lot, thanks to some parallel execution. It does only support Python 3.9+ however, and I think the performance increase depends on the use case so we should test if this also benefits ours. But that could be something for the future :-)

sjoerd-bouma, 2025-08-11:
The problem is that the generic detector currently uses a filesystem-based cache if the source is json, which is much slower than the memory-based one if a dictionary is used as a source. 

Mostly I think this is just a problem with TinyDB, and my preference is still to replace the GenericDetector implementation with a Python dictionary implementation that gets rid of TinyDB altogether - I believe @PhilippLaub has done quite a bit of work in this direction but I keep forgetting to ask him what the status is.

fschlueter, 2025-08-11:
oh if we could get rid of tinyDB that would be great but is it easy to do so?

PhilippLaub, 2025-08-11:
In  the branch `lofar/no-tinydb` I modified the base detector to use dicts instead of TinyDB. If I remember correctly, it was faster for larger detectors.

PhilippLaub, 2025-08-11:
I think there is a check for commission time.

PhilippLaub, 2026-02-19:
No, I have not worked on this for a year now, but I think it did work for LOFAR. I am not sure if the detector can be saved in an event. What I never tested was the buffering of SQL detectors.