nu-radio/NuRadioMC #593: Calculation of bandwidth in simulation.py does not work for RNO-G detector

fschlueter, 2023-11-22:
https://github.com/nu-radio/NuRadioMC/blob/15044001786cdb1856651987f536657364b076fd/NuRadioMC/simulation/simulation.py#L314-L317

The code in question
```
                self._amplification_per_channel[self._station_id][channel_id] = np.abs(filt).max()
                bandwidth = np.trapz(np.abs(filt) ** 2, ff)
                self._bandwidth_per_channel[self._station_id][channel_id] = bandwidth
                logger.status(f"bandwidth of station {self._station_id} channel {channel_id} is {bandwidth/units.MHz:.1f}MHz")
```
should calculate the bandwidth of the simulated detector (to calculate the amplitude of the thermal noise). However, as far as I understand the code, it only works when `np.abs(filt) ** 2` is 1 within the bandwidth and 0 outside. However, for the RNO-G detector (i.e., what comes our of NuRadioReco/detector/RNO_G/analog_components.py) this is not true.

fschlueter, 2023-11-22:
Solution could look like that:
```
                self.__amplification_per_channel[station_id][channel_id] = np.abs(filt).max()
                limit = np.abs(filt).max() / 100
                freqs_in_bandwidth = ff[np.abs(filt) > limit]
                bandwidth = freqs_in_bandwidth[-1] - freqs_in_bandwidth[0]
                self.__bandwidth_per_channel[station_id][channel_id] = bandwidth
                print(f"bandwidth of station channel is {bandwidth / units.MHz:.1f}MHz")
```

fschlueter, 2023-11-22:
When I am running the example `T02RunSimulation.py` with the example detector description in the same folder and plot the filter (i.e., `np.abs(file)`) I get the plot below (which explains why it worked before)

[image]

sjoerd-bouma, 2023-11-22:
I think the existing code is actually the correct thing to do? At least in terms of the formula
`Vrms = \sqrt(k bw R T)` this expression should correctly convert from temperature to Vrms, as this also scales with the amplification. I agree the term bandwidth may be somewhat confusing though.

cg-laser, 2023-11-23:
Yes, Sjoerd is correct. The code is correct, just the naming of the variable might not be ideal. Conceptually, a filter and an amplifier is the same. Both manipulate the amplitude as a function of frequency. Mathematically, and for the sake of noise level calculations, the complete S21 detector response needs to be taken into account. 
If you come up with a better name for the variable, feel free to change it. Or add a comment to avoid confusion for future people.

closed 2023-11-23 by cg-laser